sn4k3/UVtools #676: Possible XLN/DRL conversion error

tecchese, 2023-03-18:
### System

```Text
UVtools v3.12.0 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
Processor cores: 8
Memory RAM: 5.60 / 15.89 GB
Runtime: win10-x64
Framework: .NET 6.0.14
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: db9_conn_copy.pwma [Version: 516] [Class: PhotonWorkshopFile]
```

### Description of the bug

PCB Excellon file is not converted (blank image) in PCB exposure tool.

see:
[image]

GBR is ok:

[image]

Saved excellon and GBR bottom layout, then saved as PDF and converted again in GBR with Gerbview: ok...

[image]

### How to reproduce

Please try by yourself, I attached ZIP files
Thank you for your support.

### Files

[test.zip](https://github.com/sn4k3/UVtools/files/11008118/test.zip)
[untitled_2023-03-12.zip](https://github.com/sn4k3/UVtools/files/11008119/untitled_2023-03-12.zip)

sn4k3, 2023-03-18:
Problem is on drill file, it lacks of G05 == turn on drill mode and if drill is off it should not drill at all. If you insert G05 after M71 it will work.

I can latter ignore the requirement of G05 to do the holes

tecchese, 2023-03-19:
Yes, if I insert manually G05 after M71 it works but there are still problems (see video).
[Untitled.zip](https://github.com/sn4k3/UVtools/files/11010314/Untitled.zip)
The method of subtracting the holes by inverting the color for the imported Excellon file is confusing and the preview doesn't help. Also you need to add the XLN extension (used by EagleCAD) for importing Excellon files.

sn4k3, 2023-03-19:
> it works but there are still problems (see video).

What is the problem exactly? I run your drl + grl and I got the correct output as your first image of converted pdf to gbr.

> Also you need to add the XLN extension (used by EagleCAD) for importing Excellon files.

I will take a look

sn4k3, 2023-03-21:
> It might be wiser to select which layer can be inverted individually, not for all.

You can, there is a check box for inverting individual gerbers, does that work for you?

[image]

sn4k3, 2023-03-21:
Result of no inverting:

[image]

Result of inverting only one gerber:
[image]
[image]

tecchese, 2023-03-21:
For me it does not work...
I get no result when I click on the checkbox indicated by the red arrow...
this is the question...
In addition if I select a layer the size scale data disappears
[image]
Also on the top you can notice the description partially cut.

sn4k3, 2023-03-21:
For the individual invert to work, order does matter as it only inverts drawings instead of whole layer, that means where white black will be drawn, so order whites in first place is important.

> In addition if I select a layer the size scale data disappears

Looks like a UI problem, you are using the normal theme right? Can you set back to the Fluent theme?

tecchese, 2023-03-21:
When I changed theme I get this:
[image]

sn4k3, 2023-03-21:
Yes you need to restart UVtools. If unable to change and save you need to go to settings file and change there or remove the theme key

tecchese, 2023-03-22:
Ok, I selected the default Fluent Theme and everything seems to work...but it isn't true unfortunately...
see this pic:
[image]

All the invert level checkboxes (except drill level) on the right always stay active when I checked them repeatedly and there is no way to go back (i.e. to uncheck them)... very strange behaviour.

Thanks for the awesome support and patience.

sn4k3, 2023-03-22:
Fixed

closed 2023-03-22 by sn4k3