sn4k3/UVtools #683: UVTools 3.12.1 arm64 macOS fails to start

mtaumike, 2023-03-25:
### System

```Text
UVtools v3.12.1 X64
Operative system: Darwin 22.3.0 Darwin Kernel Version 22.3.0: Mon Jan 30 20:38:37 PST 2023; root:xnu-8792.81.3~2/RELEASE_ARM64_T6000 X64
Processor: Apple M1 Ultra
Processor cores: 20
Memory RAM: 23.63 / 52.75 GB
Runtime: osx.13-x64
Framework: .NET 6.0.15
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary)
    WA: 2560 x 1415    UA: 2560 x 1415

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
```

### Printer and Slicer

- Printer: N/A
- Slicer: N/A

### Description of the bug

```bash
The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x60000380cd80 {Error Domain=NSPOSIXErrorDomain Code=153 "Unknown error: 153" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}
```

### How to reproduce

Just try to launch it.

The System information is coming from the amd64/x86_64 binary running on macOS Ventura ARM

sn4k3, 2023-03-25:
You need to install it first time via: https://github.com/sn4k3/UVtools#to-auto-install-on-macos

mtaumike, 2023-03-25:
Ha, that works, but most users aren't going to have a freakin' clue about this or will be afraid of going to a command line.

Thanks for the response.

closed 2023-03-25 by mtaumike

sn4k3, 2023-03-25:
There's no other way for arm users, download and run now only works if you pay for apple developer program and sign in the app, which I don't want to do the nature of the program (Free and open)

mtaumike, 2023-03-25:
I get it completely.

I'd suggest just getting rid of the arm64 disk image from the releases page