sn4k3/UVtools #688: [Bug] SL1 "Portrait" mode not supported

Spiritdude, 2023-04-04:
### System

```Text
UVtools_linux-x64_v3.12.1.zip unpacked on Ubuntu-22.04 LTS
```

### Printer and Slicer

- Printer: Anycubic Photon Mono 4K
- Slicer: Prusa Slicer 2.4.0 & Prusa Slicer 2.5.2

### Description of the bug

When slicing to .sl1 file with "Portrait" orientation in Prusa Slicer, and then converting with UVtools.Cmd, the resulting .pwma (Mono 4K) and .pmx2 (Mono X2) are scrambled as previewed with Photon Workshop.

As I was putting the .pwma on USB stick to print on Mono 4K dry (without resin) I saw the misalignment in the UI LCD, and realized this may be the "Portrait" orientation not considered when converting .png from SL1 format.

When I selected "Landscape" in Prusa Slicer and re-exported, the resulting .pwma was correct.

.sl1 contains `prusaslicer.ini` where the setting is listed:
```
display_orientation: "portrait" vs "landscape"
```
I attach two .sl1 files, one portrait (lus_2020_p.sl1), one landscape (lus_2020_l.sl1) as reference.

### How to reproduce

UVtoolsCmd convert lus_2020_p.sl1 pwma lus_2020_p.pwma
UVtoolsCmd convert lus_2020_l.sl1 pwma lus_2020_l.pwma

### Files

[lus_2020_p.sl1.zip](https://github.com/sn4k3/UVtools/files/11145226/lus_2020_p.sl1.zip)
[lus_2020_l.sl1.zip](https://github.com/sn4k3/UVtools/files/11145227/lus_2020_l.sl1.zip)

just remove '.zip' in the filenames once downloaded.

Screenshots opening both files (see window title):

[image]
[image]

sn4k3, 2023-04-05:
**You can't use portrait in PrusaSlicer**

The reason is because PrusaSlicer somehow flip resolution and screen information when using portrait and UVtools depend on that and trigger warnings and bad images. 
I don't know if is a bug on slicer or not, but using landscape always work and doesn't matter if screen is portrait or not (That is also dependent on the viewer perspective).

As bottom line:
- Always use landscape to generate the correct information/image.
- You should use my printer profiles provided by UVtools as they are well configurated.

closed 2023-04-05 by sn4k3

Spiritdude, 2023-04-05:
> You can't use portrait in PrusaSlicer with UVtools

I'm still coding on my `mslatool`, planning to release it on github once it matured more, `sl1file.py` excerpt:
```
     if self.conf['display_orientation']=='portrait':
         im = im.rotate(90,expand=True)
```
and then you can support portrait in PrusaSlicer.

sn4k3, 2023-04-05:
I can do that but it always feels like a dirty fix

[image]

Makes no sense to have the information 
```
display_pixels_x = 3836
display_pixels_y = 2400
```

And then the image is rotated making that information not true. In general, I think that option should not exist, it serves nothing other than mess with image rotation.

Spiritdude, 2023-04-05:
I agree with the `display_orientation` setting being useless and introducing confusion - btw, it was default to "portrait" when I used Prusa Slicer to slice .sl1. 

My attitude is, make a program serve the expectation of the user (incl. me), make an effort to do so, even when incomplete or buggy information exists - and here the `display_pixels_x and y` are switched when you rotate(90), yes, think like this: you start to interpret the `display_orientation`, the rotation is already a process step in your program, hence, the `display_pixels_x/y` are switched in your program locally as a result - for whatever reason Prusa thought it was a good idea to define `display_orientation` without follow common practice with x/y notion too.

[I contacted the Prusa team about UVJ format, as soon I have a response I will ask the reason for `display_orientation`]

sn4k3, 2023-04-05:
The problem with that fix is that when you save sl1 in UVtools images will be written in the correct orientation. But parameter stills portrait, when you reopen the file in UVtools it will rotate again make it wrong.

The proper "solution" would be the rotation and set the `display_orientation` to `landscape` to fix re-open. However, this doesn't sound good to me.

Spiritdude, 2023-04-05:
There are two perspectives in action at all times:
- outside of UVtools
- inside of UVtools

You import something, then you process it and do what needs to be done internally.

You export something, then you comply with the format as defined, and yes, in the worst-case you need to rotate back again if "portrait" is wanted and you rotated at import stage in your process pipeline internally.

I see now why are so reluctant support `display_orientation` portrait mode, it breaks your internal flow and gives unnecessary overhead at export again.