sn4k3/UVtools #690: [Bug]

joehhn, 2023-04-08:
### System

```Text
UVtools v3.12.3 X64
Operative system: Microsoft Windows 10.0.19043 X64
Processor: Intel(R) Core(TM) i5-10400F CPU @ 2.90GHz
Processor cores: 12
Memory RAM: 4.47 / 15.87 GB
Runtime: win10-x64
Framework: .NET 6.0.15
AvaloniaUI: 0.10.19
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 150% (Primary) (On this)
    WA: 2560 x 1380    UA: 1707 x 920
2: 1280 x 720 @ 100%
    WA: 1280 x 680    UA: 1280 x 680

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: calibration_cubes.ctb [Version: 4] [Class: ChituboxFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn running the latest version of its firmware
- Slicer: ChituBox 1.9.4

### Description of the bug

Expected Dynamic Layer Heights to work as usual after updating UVTools, but getting this error instead.

System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.Managers.MatCacheManager.Get(UInt32 layerIndex) in UVtools.Core\Managers\MatCacheManager.cs:line 146
   at UVtools.Core.Operations.OperationDynamicLayerHeight.<>c__DisplayClass97_0.<ExecuteInternally>g__ReUseLayer|2(UInt32 layerIndex) in UVtools.Core\Operations\OperationDynamicLayerHeight.cs:line 598
   at UVtools.Core.Operations.OperationDynamicLayerHeight.ExecuteInternally(OperationProgress progress) in UVtools.Core\Operations\OperationDynamicLayerHeight.cs:line 615
   at UVtools.Core.Operations.Operation.Execute(OperationProgress progress) in UVtools.Core\Operations\Operation.cs:line 742
   at UVtools.WPF.MainWindow.<>c__DisplayClass122_0.<RunOperation>b__0() in UVtools.WPF\MainWindow.axaml.cs:line 2121

### How to reproduce

Load the attached ChituBox slice file.
Apply "dynamic layer heights" with any settings.
Result is the stacktrace quoted earlier.

NOTE: the stacktrace appears for any files I try to run, even those I sliced in CB previously.

### Files

[error example slice.zip](https://github.com/sn4k3/UVtools/files/11182455/error.example.slice.zip)

sn4k3, 2023-04-18:
I have tried with default settings and that worked for me.
Are you using different settings?

sn4k3, 2023-06-23:
Closing due lack of answer and not able to reproduce

closed 2023-06-23 by sn4k3