sn4k3/UVtools #691: [FEATURE REQUEST] Possibility to use uvtools as an "Open with..." app when right clicking an .sl1 (and other) files

pedroCX486, 2023-04-10:
### Describe the solution you'd like

When right clicking an .sl1 file, allow the possibility to select uvtools as an app to open the file with (no, not registering the extension to it as the default, just allow uvtools to have a file passed to it by context). Same would apply when dropping a file onto uvtools, detect and open the file automatically (it's passed as an arg? not sure.)

Would be amazing if it worked on macOS also, but that would be an stretch goal.

sn4k3, 2023-04-12:
Windows already have that if you install via MSI. Right-click will show "Open with UVtools".
For Linux and macOS I'm afraid it will not have that option so soon. Maybe you need to associate files manually if possible

pedroCX486, 2023-04-12:
> Windows already have that if you install via MSI. Right-click will show "Open with UVtools".
> For Linux and macOS I'm afraid it will not have that option so soon. Maybe you need to associate files manually if possible

Ah, I wasn’t aware that via MSI it could do it. I usually avoid installing things when I can.

For macOS unfortunately simply associating the files isn’t an option. I’ll have a look into what’s needed to have it open like that, my C# is rusty but worth a try.

sn4k3, 2023-04-12:
> Ah, I wasn’t aware that via MSI it could do it. I usually avoid installing things when I can.

For non-installers you can right-click file and use "Open With" locate UVtools. Once you do that it will always show on "Open With"

pedroCX486, 2023-04-13:
That works! I didn’t test early because my M1 is my main machine.

If get anywhere with the code on macOS I’ll open a PR

Cheers!

closed 2023-04-13 by pedroCX486