sn4k3/UVtools #694: [Bug] ScriptTextBoxInput, ScriptCheckBoxInput, etc rendered twice

msspad, 2023-04-25:
### System

```Text
Version 3.13.0
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

In the modal window "Scripting", input elements are shown twice:
[image]

DoubleInput.cs:
```using System;
using UVtools.Core;
using UVtools.Core.Extensions;
using UVtools.Core.Scripting;

namespace UVtools.ScriptSample;

public class DoubledInputs : ScriptGlobals
{ 
    private ScriptTextBoxInput abcInput = new() {
        Label = "abc",
        Value = ""
    };

    public void ScriptInit() {
        Script.Name = "abc";
        Script.UserInputs.Add(abcInput);
    }

    public string? ScriptValidate() {
        return "";
    }

    public bool ScriptExecute() {
        return !Progress.Token.IsCancellationRequested;
    }
}
```

### How to reproduce

Load DoubleInput.cs and run the script. Reopen the window "Scripting" again and in 80% of the cases, the input elements get rendered twice, see screenshot. So this bug does not always pop up.

sn4k3, 2023-04-25:
I can confirm the problem.
This happens when using the restore session setting
[image]

sn4k3, 2023-04-27:
Fixed

closed 2023-04-27 by sn4k3