sn4k3/UVtools #695: [FEATURE REQUEST] Open the app on the MAIN monitor instead of whatever monitor the desktop icon is on

ScienceofSpock, 2023-04-28:
### Is your feature request related to a problem?

I have 3 monitors. I keep my 3D printing-related app icons on my desktop on the right monitor. Whenever I launch UVTools, it ALWAYS starts on the right monitor. Everything else behaves properly and opens on my MAIN monitor, but not UVTools. It launches on whichever monitor the icon is on.

### Describe the solution you'd like

I would like it to open on the main monitor like virtually every other application in existence.

sn4k3, 2023-04-28:
That is possible by using this combination of settings: 

[image]

After save those settings, move UVtools to the screen you want, optionally maximize the window. Close UVtools and reopen, you will now have it start on that screen

ScienceofSpock, 2023-04-28:
Except that doesn't work either, it starts BASICALLY on the right monitor, but it is not technically maximized and starts about 10 pixels to the right and I have to move it anyway.
[image]

sn4k3, 2023-04-28:
It works on my setup, using two different monitors.
Are you using scaling under display settings?

ScienceofSpock, 2023-04-28:
No, I am not using scaling.

ScienceofSpock, 2023-04-28:
I guess my real question is why does it open on the monitor where the icon is? I don't think I have ever seen desktop icon location used as a basis for which monitor to open an application on before.

sn4k3, 2023-04-28:
No clue, the UI framework is what control that and there is no option to select starting screen as I see. Documentation says it opens on the screen where mouse is when you open the app.
I will try to combine the maximized window with other two settings to solve that.

ScienceofSpock, 2023-04-28:
I had thought about mouse position too, and maybe I'm not fast enough, but it doesn't seem to matter. The only thing that seems to matter is where the icon is. If I put it on the left monitor, it opens on the left monitor. If I put it on the right one, it opens there.

At the end of the day, it's just a mild annoyance. Thanks for looking.

ScienceofSpock, 2023-04-28:
Well, I just figured out a workaround. Using the settings you suggested earlier puts the window on the correct monitor, but as I mentioned, it was off by like 10 pixels. I just looked at the shortcut, and if you change the "Run" setting from the default of "Normal Window" to "Maximized" then it works! See, I had the window maximized before, but it seems you have to actually change the setting in the shortcut, at least in my situation.

closed 2023-04-29 by ScienceofSpock