sn4k3/UVtools #696: [Bug] Error Opening Sliced File for Halot Mage

cbanderson22, 2023-04-30:
### System

```Text
UVtools v3.13.1 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 9 7900X 12-Core Processor            
Processor cores: 24
Memory RAM: 10.60 / 31.14 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.19
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1440    UA: 2560 x 1440
2: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080
3: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Creality Halot Mage
- Slicer: Chitubox Pro V1.4.0Beta2

### Description of the bug

Upon trying to load a sliced file for this new printer, the following error occurs. I'm assuming it has to do with the new file type this printer uses.
[image]

### How to reproduce

Download new beta of Chitubox and slice a file for the Halot Mage printer. Then load into UVtools.

sn4k3, 2023-04-30:
I don't have Chitubox Pro, please attach a file

cbanderson22, 2023-04-30:
Sure thing, here's a small sliced file.
[Halot Test File.zip](https://github.com/sn4k3/UVtools/files/11361750/Halot.Test.File.zip)

sn4k3, 2023-04-30:
That ctb file have 4 extra bytes on the end, making UVtools not to load the file.
Try to slice another file and upload here to see if those 4 bytes are the same or not.
Also try to dry print this file without those 4 bytes and see if printer start the print: [Halot Test File_no4bytes.zip](https://github.com/sn4k3/UVtools/files/11361777/Halot.Test.File_no4bytes.zip)

cbanderson22, 2023-04-30:
So here is another sliced file done the same way. Also I can confirm that your modified file fires up on the printer correctly.
[Halot Test Cube.zip](https://github.com/sn4k3/UVtools/files/11361841/Halot.Test.Cube.zip)

sn4k3, 2023-04-30:
The test cube file also has the extra 4 bytes that are not equal to another file, so it's not a constant.

> New beta version adding extra bytes for reasons?

I don't know, as that is not required anyway maybe it's to break UVtools 🤣

As I don't understand this change nor see a Version change, I will not read those bytes. I may change the way checksum is obtained in order to bypass that problem but for now I will keep as it is until non-beta cames out.

To solve this issue yourself, you remove the last four bytes from your files. Install a free hex editor, go to end and delete 4 bytes, save and it can now load.

cbanderson22, 2023-04-30:
Fair enough, I can work with that solution. Thanks for the help!