sn4k3/UVtools #701: [Bug] Anycubic Photon Mono X - Wait before cure global setting not working

chakflying, 2023-05-10:
### System

```Text
UVtools v3.13.3 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 5 5600X 6-Core Processor             
Processor cores: 12
Memory RAM: 23.52 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.19
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 100% (Primary) (On this)
    WA: 3440 x 1440    UA: 3440 x 1440

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: battery-box-03.pwmx [Version: 1] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic Photon Mono X
- Slicer: PrusaSlicer

### Description of the bug

#533 Mentioned that the per-layer "wait before cure" does not work, but it seems like the global setting "WaitTimeBeforeCure" setting cannot be edited either. This setting can be set in Lychee Slicer and PrusaSlicer, but changing it in UVTools would revert the value back to 2s.

### How to reproduce

1. Open a .pwmx file sliced by Prusaslicer
2. edit the global value "WaitTimeBeforeCure" to 20s
3. save file
4. Observe the value "WaitTimeBeforeCure" return to 2s

sn4k3, 2023-05-10:
Settings are kept for me. Are you auto applying wait time before cure suggestion?

chakflying, 2023-05-10:
Yep that was it, I must have changed the setting a long time ago and forgot about it. A bit strange since I assume it would only apply suggestions after it has checked for issues.

closed 2023-05-10 by chakflying

sn4k3, 2023-05-10:
Suggestions are independent from issues and vice versa.