sn4k3/UVtools #702: No "Do not perform lifting sequence for layers with same Z positioning" when generating calibration file for Anycubic M3 Premium

fcollingwood, 2023-05-11:
### System

```Text
UVtools v3.13.3 X64
Operative system: Microsoft Windows 10.0.22000 X64
Processor: Intel(R) Xeon(R) W-3223 CPU @ 3.50GHz
Processor cores: 16
Memory RAM: 148.56 / 255.65 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.19
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 100% (Primary) (On this)
    WA: 3440 x 1392    UA: 3440 x 1392
2: 3440 x 1440 @ 100%
    WA: 3440 x 1392    UA: 3440 x 1392

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UV Tools RERF.pm3r [Version: 517] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic Photon M3 Premium
- Slicer: N/A

### Description of the bug

Expecting to be able to select "Do not perform lifting sequence for layers with same Z positioning" in the Multiple Exposures section, but it's not there - not even there but greyed out, just completely missing

### How to reproduce

Try to create a calibration file for an Anycubic M3 Premium

sn4k3, 2023-05-11:
It's there for me, try to scroll the section down.

fcollingwood, 2023-05-11:
Nope. Not there

[image]

sn4k3, 2023-05-11:
Ah yes, sorry I was looking for other option.
That option is gone long time ago. To get same result set 0mm on 'Lift Height' option below. But that won't work for most formats, and that's why default is 0.10mm. So far only gcode printers can benefit from that. All other formats doesn't like the zero value

fcollingwood, 2023-05-11:
That's a great pity, because non-gcode printers had no issues with the "Do not perform lifting sequence for layers with same Z positioning" option - I generated multiple exposure tests for my Phrozen Mighty 4Ks just a few weeks ago on an older version. Looks like I need to downgrade my installation back to what I was using