sn4k3/UVtools #703: [Bug] System.IO.InvalidDataException: End of Central Directory record could not be found.

Hirsu, 2023-05-11:
### System

```Text
UVtools v3.13.3 X64
Operative system: Microsoft Windows 10.0.22000 X64
Processor: 12th Gen Intel(R) Core(TM) i7-12700K
Processor cores: 20
Memory RAM: 5.88 / 31.79 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.19
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1392    UA: 2560 x 1392
2: 1920 x 1080 @ 100%
    WA: 1920 x 1032    UA: 1920 x 1032

Path:       C:\_Utilities\UVTools\
Executable: C:\_Utilities\UVTools\UVtools.exe
```

### Printer and Slicer

- Printer: Uniformation GKTwo
- Slicer: Uniformation Slicer

### Description of the bug

After a fresh installation of UV Tools, I get the following error when I open any .jxs file:

> System.IO.InvalidDataException: End of Central Directory record could not be found.
>    at System.IO.Compression.ZipArchive.ReadEndOfCentralDirectory()
>    at System.IO.Compression.ZipArchive..ctor(Stream stream, ZipArchiveMode mode, Boolean leaveOpen, Encoding entryNameEncoding)
>    at System.IO.Compression.ZipFile.Open(String archiveFileName, ZipArchiveMode mode, Encoding entryNameEncoding)
>    at System.IO.Compression.ZipFile.Open(String archiveFileName, ZipArchiveMode mode)
>    at UVtools.Core.FileFormats.JXSFile.DecodeInternally(OperationProgress progress) in UVtools.Core\FileFormats\JXSFile.cs:line 424
>    at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 3606
>    at UVtools.WPF.MainWindow.<>c__DisplayClass110_0.<ProcessFile>b__1() in UVtools.WPF\MainWindow.axaml.cs:line 1343

I get this for any file. Tried uninstalling and reinstalling, different files, different installation locations. No dice so far.
Any help would be greatly appreciated, thanks!

P.S.: Attached a sample .jxs file
[eiffel_final.zip](https://github.com/sn4k3/UVtools/files/11449309/eiffel_final.zip)
 I was trying to open, just as a reference. Issues persists on other files.

### How to reproduce

1. Open UV Tools
2. File -> Open -> Select  any .jxs file
3. Error as described above

sn4k3, 2023-05-11:
gktwo messed the format by using same extension, however .jxs from gkone is a zip file, .jxs from gktwo is a CTB file... Just rename .jxs to .ctb and it will work.

Hirsu, 2023-05-11:
> gktwo messed the format by using same extension, however .jxs from gkone is a zip file, .jxs from gktwo is a CTB file... Just rename .jxs to .ctb and it will work.

Good to know, thank you!

I'll post this information in their Facebook group, so other Uniformation users can continue using your awesome project as well.

closed 2023-05-13 by Hirsu

Pussy123a, 2024-04-14:
Hello, I had the same Issue, then I renamed .jxs to .ctb and now I get this message:

System.UnauthorizedAccessException: Access to the path is denied.
   at System.IO.FileSystem.MoveFile(String sourceFullPath, String destFullPath, Boolean overwrite)
   at System.IO.File.Move(String sourceFileName, String destFileName, Boolean overwrite)
   at UVtools.Core.FileFormats.FileFormat.Encode(String fileFullPath, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 4133
   at UVtools.Core.FileFormats.FileFormat.SaveAs(String filePath, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 5678
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in UVtools.Core\FileFormats\FileFormat.cs:line 4221
   at UVtools.UI.MainWindow.<>c__DisplayClass97_0.<ProcessFile>b__2() in UVtools.UI\MainWindow.axaml.cs:line 1432

UVtools 4.3.0 X64
Operative system: Microsoft Windows 11
Printer: Uniformation GKTwo
Slicer: Uniformation Slicer