sn4k3/UVtools #707: [Bug]

BblackmilkK, 2023-05-14:
### System

```Text
UVtools v3.13.3 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
Processor cores: 16
Memory RAM: 5.79 / 15.71 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: avatar Full_B.pwmx [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: anycubic photon mono x
- Slicer: chitubox pro

### Description of the bug

3.11 i good anad no error
 but 3.11.0......3.13.3 (Version 3.11.0 through 3.13.3) it have bug. When you make changes to the file (with Pixel Editor) and save it, it doesn't open again.

In no case do files saved after version 3.11.0 open. (Included in the 3.13.3 final release.)

### How to reproduce

when i try to open a saved file :
[image]

### Files

[image]
[image]

BblackmilkK, 2023-05-14:
i uploaded files : https://we.tl/t-6DoYXLOPkQ

sn4k3, 2023-05-14:
Will be fixed on next patch

BblackmilkK, 2023-05-14:
> Will be fixed on next patch

thank you

sn4k3, 2023-05-21:
Fixed

closed 2023-05-21 by sn4k3

BblackmilkK, 2023-05-21:
> Fixed

thank you i dowloaded. and yes it is fixed thank you so much