sn4k3/UVtools #717: [Bug] Exported .goo file from Lychee 5.2 - "Error opening the file" - Decoded RLE for layer ... is corrupted, ...

Cainam-git, 2023-05-29:
### System

```Text
UVtools v3.13.4 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz
Processor cores: 8
Memory RAM: 9.03 / 31.94 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 1080 x 1920 @ 100%
    WA: 1080 x 1920    UA: 1080 x 1920
2: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1410    UA: 2560 x 1410
3: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Elegoo Mars 4 Max
- Slicer: Lychee Slicer 5.2

### Description of the bug

**What is the problem?**
A .goo file exported from Lychee Slicer 5.2 seemingly can't be imported into UVTools. When I switch to a different printer profile (e.g.: Elegoo Mars 3 Pro) and export a different file type (e.g. .ctb) the Lychee + UVTools combo works like a dream again.

The problem is Elegoo shifted to the (much more friendlier) .goo file extension with the Mars 4 Max. There are no options to export a different file type.

**What did you expect?**
The .goo file to open in UVTools for analysis. Either via the "File => Open" way or directly from within Lychee Slicer with the "Open with UVTools" button.

**Additional information**
It doesn't matter if you do anything in the slicer. This can be reproduced by grabbing a random object (e.g. a 'sphere') and immediately going to the "export" tab.

PS: I'm not sure if this is a [Bug] or a [Feature Request]. Since recent patch notes mentioned "support for .goo file format" somewhere I assumed a bug. I'm sorry if I got it mangled up.

### How to reproduce

1. Select the Mars 4 Max printer profile in Lychee Slicer 5.2
2. Under "export slices", export the slice to a .goo file
3. Open the .goo file in UVTools 3.13.4
4. Bask in the glory of an import error

[image]
[image]
[image]

sn4k3, 2023-05-29:
Duplicate of https://github.com/sn4k3/UVtools/discussions/710
This should be on lychee side, please contact them to review this problem.

Cainam-git, 2023-05-29:
My sincere apologies, I've contacted Lychee!

Thank you for making this amazing piece of software!

closed 2023-05-29 by Cainam-git