sn4k3/UVtools #719: [Bug]

dpalay, 2023-05-31:
### System

```Text
UVtools v3.13.4 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-5930K CPU @ 3.50GHz
Processor cores: 12
Memory RAM: 22.15 / 31.90 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080
2: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1400    UA: 2560 x 1400
3: 1600 x 1200 @ 100%
    WA: 1600 x 1200    UA: 1600 x 1200

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Anycubic Photon Ultra
- Slicer: Lychee Slicer - 5.2.101

### Description of the bug

After exporting my scene as a .dlp file, I attempted to load it into UVTools. 
UVT started scanning through the file and almost immediately provided this stack trace.

```
(1) System.IO.FileLoadException: Image ended short: 914661, expecting: 921600
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.DecodePW0() in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 853
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.Decode(Boolean consumeData) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 644
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass203_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1955
   at System.Threading.Tasks.Parallel.<>c__DisplayClass32_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
---------------
(2) System.IO.FileLoadException: Image ran off the end: 920334+ 1271 = 921605, expecting: 921600
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.DecodePW0() in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 836
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.Decode(Boolean consumeData) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 644
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass203_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1955
   at System.Threading.Tasks.Parallel.<>c__DisplayClass32_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
---------------
(3) System.IO.FileLoadException: Image ended short: 921420, expecting: 921600
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.DecodePW0() in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 853
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.Decode(Boolean consumeData) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 644
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass203_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1955
   at System.Threading.Tasks.Parallel.<>c__DisplayClass32_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
---------------
(4) System.IO.FileLoadException: Image ran off the end: 920340+ 1271 = 921611, expecting: 921600
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.DecodePW0() in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 836
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.Decode(Boolean consumeData) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 644
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass203_0.<DecodeInternally>b__1(Int32 layerIndex) in UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1955
[... 90 lines cut ...]
   at System.Threading.Tasks.Parallel.<>c__DisplayClass32_0`2.<ForEachWorker>b__0(Int32 i)
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
```

I tried re-exporting the scene to different locations (network, local) and with different settings (Anti-aliasing on or off) and each time produced a similar stack trace of the image running short or long.

### How to reproduce

1. Open the .stl in Lychee
2. Rotate & fit object to plate
3. Make sure printer is set to Anycubic Photon Ultra DLP with the following settings
[image]
4. export slices to file (as .dlp)
[image]
5.  try to open exported file in UV Tools

Alternatively, 
1. Try to open any of the .dlp files in UV Tools

### Files

https://drive.google.com/file/d/1YtPg_ztANxOreUf8Qm5hs9HMoRzri5ZA/view?usp=sharing

sn4k3, 2023-05-31:
This should be on lychee side. Exporting to pwmx works and it use the exact same format as dlp and all other anycubic printers.
Try to reach lychee team about this problem. I will inform them too.

Tibus, 2023-05-31:
A bug in the .dlp format has been introduced in the new 5.2.1 of Lychee Slicer. It will be fixed in the 5.2.2. Sorry about that!