sn4k3/UVtools #722: [Bug] 2) Bottom Retract Speed & 2) Retract Speed incorrectly labelled slow speed.

JohnScottUK, 2023-06-06:
### System

```Text
UVtools v3.14.1 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Mon Apr 24 21:10:53 PDT 2023; root:xnu-8020.240.18.701.5~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i7-4980HQ CPU @ 2.80GHz
Processor cores: 8
Memory RAM: 7.21 / 15.83 GB
Runtime: osx.12-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 200% (Primary)
    WA: 1440 x 806    UA: 720 x 403

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: LycheeSpeed1.pm3r [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic Photon M3 Premium
- Slicer: Anycubic Photon Workshop, Chitubox or Lychee

### Description of the bug

When looking at the Tools -> Edit Print Parameters, the 2) Bottom Retract Speed and 2) Retract Speed are labelled in the pop-up as "Slow retract speed of the [bottom/normal] layers (TMSC)"

Using long lift distances (30 & 50 mm) and large speeds (240 & 480mm/min) it can be seen that the 2) Bottom retract Speed and 2) Retract speed values occur on the first retract stage, so are normally the fast speeds before the "Bottom Retract Speed" and "Retract Speed" are used as the second retract stage (the slow stage).

### How to reproduce

In Anycubic PhotonWorkshop:
1. Change Slicer settings:

Control Type: Advanced
Z Lift Step [0] Speed: 240
Z Lift Step [1] Speed: 480
Z Retract Step[0] Speed: 240
Z Retract Step[1] Speed: 480
for Bottom and Normal Layers

Change heights to 10x their default values (e.g. 30 & 50 mm). Change expsure times to 0.25s (to speed up test).

Run test with empty VAT and no build plate to see speeds.

Cycle is:
Lift Step[0]
Lift Step [1]
Retract Step[1]
Retract Step[0]

In Chitubox:
Change the same except Order on screen is:
Bottom Lift Speed: 240 & 480
Lift Speed: 240 & 480
Bottom Retract Speed is 480 & 240
Retract Speed is 480 & 240

Change other settings for similar dry run.
Run test empty VAT and no build plate.
Observe Cycle:

Lift is 240 (Slow) then 480 (Fast)
Retract is 480 (Fast) then 240 (Slow).

Now check print settings in UVTools. Observe that 2) Bottom Retract Speed and 2)Retract Speed are set to 480mm/min (the fast settings).

sn4k3, 2023-06-07:
I tried to slice a file and looks like they are on correct place:

[image]

Order is as following:

**Lift1 (Slow) -> Lift2 (Fast) -> Retract1 (Fast) -> Retract2 (Slow)**

So you are setting Retract[0&1] in reverse order

```
Z Retract Step[0] Speed: 240
Z Retract Step[1] Speed: 480

Should be set as:

Z Retract Step[0] Speed: 480
Z Retract Step[1] Speed: 240
```

JohnScottUK, 2023-06-07:
I will double check my findings but that's not what I observe when I actually dry run the sliced file (on M3 Premium). 

I observe Lift0 (Slow) -> Lift1 (Fast) -> **Retract1 (Fast)** -> **Retract0 (Slow)**

I set the 0 to slow speed (240mm/min) and the 1 to fast speed (480) with a large lift/retract heights (30mm and 50mm).

I will try again using lift[0] and retract[0] at 60mm/min and lift[1]/retract[1] at 480mm/min. Will video dry run to share observation.

JohnScottUK, 2023-06-07:
It may be a PM3R format-specific thing. I will setup and record a dry run for you.

I note the default settings are:
Bottom Lift[0]: 2mm @ 2mm/s
**Bottom Retract[0]: 4mm/s**
Bottom Lift[1]: 5mm @ 4mm/s
**Bottom Retract[1]: 6mm/s**

Normal Lift[0]: 3mm @ 2mm/s
**Normal Retract[0]: 4mm/s**
Normal Lift[1]: 3mm @ 4mm/s
**Normal Retract[1]: 6mm/s**

So retract[1] is faster than retract[0].

When I do the same in Lychee & Chitubox, the on-screen fields are reversed as expected, but they produce PM3R formats with Lift/Retract[0] being close to the FEP and Lift/Retract[1] being higher.

Also Lychee https://mango3d.io/changelog-archive/ 3.7 mentions "Anycubic Mono X 6K and Mono 4K: improved profile and two-stage motion **(lowering speed were inversed)**" which might imply its Anycubic doing something different in their formats to everyone else.

JohnScottUK, 2023-06-07:
Here's the dry run showing the sequence:
lift0 then lift1
retract1 then retract0.

https://www.youtube.com/watch?v=XZip6xuiVD4

sn4k3, 2023-06-07:
Check last release

closed 2023-06-11 by sn4k3