sn4k3/UVtools #724: Latest version destroys my models[Bug]

BrianGeorgeH, 2023-06-09:
### System

```Text
UVtools v3.14.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-7820HK CPU @ 2.90GHz
Processor cores: 8
Memory RAM: 9.90 / 31.96 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 100% (Primary) (On this)
    WA: 3840 x 2160    UA: 3840 x 2160

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Sizzling Bigery-Jaban.pwma [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: 
- Slicer: 
Photon Mono 4K and 6K
Lycee Slicer (Pro version)

### Description of the bug

Nothing works properly anymore!!
1. I see overhangs as if comletely adequate supports from lychee slicer only provide a couple of pixels support  to base of models. Previous UVTools version showed no overhangs same file....
2. When I search for issues and it finds islands they are not shown on the issues pixel level display. The display now shows huge properly supported sections of my models in light blue instead of black as before
3. When I selected just to repair a handful of 1 pixel islands with solid 1 pixel below the gap the repair totally destroys my model. This never happened in the other releases
4. Previous fully compliant models repaired by UVTools and that print properly now load showing lots of non-existant problems

### How to reproduce

Just load the model I attach - previous version of UVTools showed a few irrelevant overhangs now its massive ones. The previous version showed the same 5 1 pixel islands (1 layer deep) and solved them. The new version completely wrecks the model when even limited to just 1 pixel island and 1 level below actions! The model now shows all the details nearly as blue pixels not black as should be (they have full suport below them). Islands when selected show nothing at all no matter what options are selected.

I produce complex models and rely 100% on UVtools since 2 years with perfect results. Now I cannot use my printers none of the slicer tools are good enough for the models I create without your tool. I have given you a simple example most of my models are large complex ones... but if such a simple one fails then it seems hopeless.

### Files

[UVToolsBug.zip](https://github.com/sn4k3/UVtools/files/11707947/UVToolsBug.zip)

BrianGeorgeH, 2023-06-09:
Here I show the overhangs - in previous version their were none! Now supports are providing wierd cross like support!!
[image]
[image]

BrianGeorgeH, 2023-06-09:
Here I show a layer which has 100% support below it but only a tiny part shows its supported (black) the rest is a wierd blue...
There is an identical layer below and above this one so this is bizarre...
[image]

sn4k3, 2023-06-09:
When you say "The previous version" can you tell which is? Because nothing changed long time ago for issue detection/repair, made me think you were using a very old version. Blue are suction cups.

If I ignore overhangs which are not reparable by UVtools I got this:

[image]

From there only 1px islands will be "repaired" all others will be untouch.
Suction cup will attempt to drill a small hole, but that will not really affect your model.
The cross you see from overhangs are just what detection sees but it's not your real model, If you distoggle "Issues" button near "Outline" it will show you the real model without issues colors

[image]

None of these are really the problem. The real problem is that you are run the auto repair model, which attempts to close small areas by default (UVtools does that since ever) and that lead to:

[image]

So your problem is not islands nor overhangs nor suction cups nor resin traps BUT a gap closing on repair pass:

[image]

If you set that setting to 0, it will not destroy your detail.
For such details you should never run auto-repair with gap closing nor noise removal. But islands and other issues should be addressed manually by you. 

UVtools is a tool to help and use in combination with user, but never to replace user with a whole automatic thing.

Also, I recommend you to study the different issues types, why they apear, the thresholds, how they are fixed, etc.
As far I can tell by run your file all the displayed issues are real given the threshold, if they represent a real problem well it depends, I can tell you that the suction cups you have will not represent a problem, nor the small 1px islands nor the resin trap. The big islands/overhangs it will depend on your settings and resin, they are detected because does not meet the supportive mass below.

sn4k3, 2023-06-09:
When you say "The previous version" can you tell which is? Because nothing changed long time ago for issue detection/repair, made me think you were using a very old version. Blue are suction cups.

If I ignore overhangs which are not reparable by UVtools I got this:

[image]

From there only 1px islands will be "repaired" all others will be untouch.
Suction cup will attempt to drill a small hole, but that will not really affect your model.
The cross you see from overhangs are just what detection sees but it's not your real model, If you distoggle "Issues" button near "Outline" it will show you the real model without issues colors

[image]

None of these are really the problem. The real problem is that you are run the auto repair model, which attempts to close small areas by default (UVtools does that since ever) and that lead to:

[image]

So your problem is not islands nor overhangs nor suction cups nor resin traps BUT a gap closing on repair pass:

[image]

If you set that setting to 0, it will not destroy your detail.
For such details you should never run auto-repair with gap closing nor noise removal. But islands and other issues should be addressed manually by you. 

UVtools is a tool to help and use in combination with user, but never to replace user with a whole automatic thing.

Also I recommend you to study the different issues types, why they apear, the thresholds, how they are fixed, etc.
As far I can tell by run your file all the displayed issues are real given the threshold, if they represent a real problem well it depends.

PS: All colors you see (that are not black and white) are virtual drawn, they are not actually used by the printer and can be toggle off on the UI while inspecting.

BrianGeorgeH, 2023-06-09:
Thanks my version was from 2022 so old. How do I turn off the blue colours on the layer view?

BrianGeorgeH, 2023-06-09:
Thanks all is good... Sorry for bothering you.

closed 2023-06-09 by BrianGeorgeH