sn4k3/UVtools #725: [FEATURE REQUEST] Add setting in preferences to configure default file name

jbmay, 2023-06-10:
### Is your feature request related to a problem?

Currently after you use UVTools to modify a slice, when you go to save the file it has the default file name set to the original file name with "_copy" appended to it.

[image]

### Describe the solution you'd like

It would be nice if there was a field in the settings to allow users to configure how they want the default file name to be formatted when they save a file. For example I would like to name the file something like the following:

[image]

Where "5u_2.5s_J3D's Boxes of Calibration" was the original name of the file, "00h49m" is the calculated print time of the file, and "ready" is a literal that I want on every slice file to indicate it has gone through UV Tools and is ready to be printed.

This could be done by providing a text field in the preferences that provides some variables that users can use to define their desired default file names. An example of how a user might use this field could look like `${ORIGINAL_FILENAME}_${PRINT_TIME}_ready` where anything inside a `${}` is a variable that UVTools allows to be referenced in the field and anything outside of those is treated as a literal.

sn4k3, 2023-06-11:
I've added this with some more goodies:

[image]

All public properties inside [FileFormat.cs](https://github.com/sn4k3/UVtools/blob/master/UVtools.Core/FileFormats/FileFormat.cs) can be used

sn4k3, 2023-06-11:
Implemented, please try it

closed 2023-06-11 by sn4k3

jbmay, 2023-06-11:
Awesome thanks! Can't believe how quick you implement requested features.

jbmay, 2023-06-11:
Tested it and confirmed it does what I was looking for!