sn4k3/UVtools #728: [Bug]

Hazoooka, 2023-06-19:
### System

```Text
UVtools v3.14.0 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Mon Apr 24 21:10:53 PDT 2023; root:xnu-8020.240.18.701.5~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
Processor cores: 24
Memory RAM: 10.09 / 24.16 GB
Runtime: osx.12-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Users/witos/Applications/UVtools.app/Contents/MacOS/
Executable: /Users/witos/Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Description of the bug

UVtools does not remember any changes to the preferences, or the recent files opened.
It's the only application that fails in my system, so it's not a system fault.

What I have discovered that it's not a native Mac OS build but frankenstined windows compilation.
That's the most possible cause of problems.

Please make a native Mac OS build.

### How to reproduce

Make a change to the preference file

sn4k3, 2023-06-19:
> What I have discovered that it's not a native Mac OS build but frankenstined windows compilation.
That's the most possible cause of problems.

Please make a native Mac OS build.

You running native macos build. There's no way to run windows apps on macos.

I found that the setting folders UVtools uses no longer exists on Monterey. Will fix on next build.

sn4k3, 2023-06-19:
> From what I have investigated in the config files a lot of the libraries are described as M$ ones.

And that automatically means the program is not native? Microsoft as any other publisher is free to develop for macOS. So that assumption means nothing if an app is native or not.
UVtools uses .NET framework which compiles natively for macOS, both x64 and arm64. 
You can check the process tree and will find that UVtools have no parent nor child processes.
With UVtools you are getting native performance of your host, for example, arm64 users (m1/m2) are beating most of Windows users in raw performance with a big margin.
That been said, your request for a native build is already satisfied.

> M$ way in this case, or basing on the M$ technology, which by default means problems. And since UVtools is the only program failing out of 6 I’m using it is a logical explanation.

I already told where the problem is, it's nothing related to microsoft nor UVtools. It's the simple fact the recent macOS removed a folder that was used by UVtools and other programs to save data, it is: `$HOME/.local/share`. With that folder no longer part of the OS is also no longer possible to create it by user since dot started folders are reserved to system and so UVtools is not able to create it.

On next release I will use different folder, that will be: `$HOME/Library/Application Support`

sn4k3, 2023-06-19:
How settings are hidden?

Hazoooka, 2023-06-19:
$HOME/.local/share

sn4k3, 2023-06-19:
I don't know how Apple finder works, but I guess by default it will not show under search if you have to not show invisible folders which you can turn ON by: `Cmd ⌘ Shift ⇧ .`

Here's the folder tree: 

[image]

Nevertheless it will never show on your Monterey because apple decided to remove that folder from the system. I don't know why nor excepting that anyway.

UVtools also allow you to open the folder directly if you go to Help -> Open settings folder, however as that folder don't exists it will not open.

Hazoooka, 2023-06-19:
The best would be to dump the .NET technology and make it the proper way, so it can be compiled for three systems with regards of each system's native build principles.
Do not cut corners, it always shows sooner or later.
Taking the easy ways is the worst way of doing things.
Make it properly.
In software world quality matters.
Much to learn but it will pay off one day.

Kind regards,
Witold

> So please, put some work in it, and make a native compilation for Mac OS.
>

sn4k3, 2023-06-19:
I will not be arguing with you further, but the things you told so far are not as you think.
You come here to report the real issue: "UVtools does not remember any changes to the preferences, or the recent files opened." but end in trashing UVtools and the running tech behind (MS .NET). If you unhappy with it you are free to remove from your system, or if you have such strong knowledge convert it to a pure macOS app using the apple framework, as the project is open-source everyone can fork, adapt and modify. I don't have any intention nor want to dump the net framework and more than triplicate the project complexity/work which is all on me and a hobby, given for free with almost 0 return from it. Today developing an app for different systems using each system framework is nothing more than a waste of time, most don't need the extra performance, that's why abstraction framework exists, and .NET is one of the best. Also, sorry to say but when it comes to developing, apple is the most painful to deal with as you have to pay their license to publish the app which caps free and open developers.

As a last note, and UVtools being a native macOS app, it uses the Apple Framework via the .NET abstraction.
If you inquiry all dylib (native macOS libs) of UVtools you will find that they all use the Apple Framework which UVtools ends using after all, example of the application core:

[image]

Would dump .NET in prior of pure C++ with the Apple Framework give better performance? Yes, but minimal. Because almost all operations UVtools performs on sliced files are made by OpenCV which is natively compiled and external called, those is what gives UVtools its performance. If you not yet observed that UVtools is one very optimized program, blazing fast for what it does and stable, then I don't know what to say, perhaps install native Photoshop made with pure Apple Framework, open 1000 4k images on it and attempt to batch apply a simple filter to them at "same time", maybe that way you give more value to this software and the technology running behind it.
You can also open Help -> benchmark tool and compare the results with my Windows i9-9900k CPU, probably to realize that the .NET framework is not that bad running on macOS as it will probably beat my system on certain profiles.

___

The true scope of the problem: "UVtools does not remember settings on Monterey and above" will be solved by simply change the location where the files are being saved. All other stuff is unrelated.

sn4k3, 2023-06-21:
Fixed

closed 2023-06-21 by sn4k3

Hazoooka, 2023-06-21:
Thank you!

Witold