sn4k3/UVtools #729: Export to *.cxdlpv4 Dord Not Work[Bug]

kenweise, 2023-06-19:
### System

```Text
UVtools v3.14.3 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 7 5800H with Radeon Graphics         
Processor cores: 16
Memory RAM: 11.89 / 13.86 GB
Runtime: win10-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100%
    WA: 1920 x 1032    UA: 1920 x 1032
2: 2560 x 1080 @ 100% (Primary) (On this)
    WA: 2560 x 1032    UA: 2560 x 1032

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_Elephant Foot.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: Creality Halot Mage Pro
- Slicer: Chitubox Pro

### Description of the bug

Problem: When converting a calibration file to the CXDLP4 format required by the Mage Pro, UVTools produces an error. The Mage Pro does not accept other Chitbox outputs like CTB, CTB Encrypted, etc.

### How to reproduce

Steps to Reproduce

- Create a new Calibration item
- Select File->Convert To->Creality CXDLP
- Select a Version. All selections give the same error
- Select a filename

Expected Result: File created that can be used in the printer

Actual Result: Error
[image]

sn4k3, 2023-06-19:
UVtools does not support cxdlpv4 format nor offer the possibility to export for such format.
cxdlp != cxdlpv4

kenweise, 2023-06-19:
cxdlp format selection is what causes the error.

sn4k3, 2023-06-20:
CXDLP format have a field that requires the correct printer model to be set, that model if not present on name is extracted from the resolution spec, if unable to match the resolution from UVtools compatible printer's database that error will throw. As that printer (Creality Halot Mage Pro) is not compatible with cxdlp format it doesn't allow you to convert to that format.

However, you can trick UVtools and set the model via Terminal, that will make the convertion to work but printer will not accept that file.
If you want to try, go to File -> Terminal, and write: `SlicerFile.MachineName = "CL-XX";` 
I don't know the exact machine code model of Mage Pro

closed 2023-06-23 by sn4k3