sn4k3/UVtools #730: [Bug]

Hazoooka, 2023-06-19:
### System

```Text
UVtools v3.14.0 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Mon Apr 24 21:10:53 PDT 2023; root:xnu-8020.240.18.701.5~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
Processor cores: 24
Memory RAM: 8.45 / 24.27 GB
Runtime: osx.12-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Users/witos/Applications/UVtools.app/Contents/MacOS/
Executable: /Users/witos/Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Description of the bug

Edit print parameters > change settings per layer range
Saving the changes makes the parameter's window to disappear.
Then, after tedious process of opening it again and again for every layer, the layer that is being edited is random.
Sometimes the next layer after the last edited is available for editing, sometimes the last layer, sometimes the first one is available.

### How to reproduce

Load the CTB file.
Open the "Edit print parameters window"
select "change settings per layer range"
change any parameter for layer 1 and save it
Open the "Edit print parameters window" again and observe which layer is automatically opened for editing.

Please make it organised so the user does not have to shuffle through hundreds of layers to get the next in line for editing.

sn4k3, 2023-06-20:
> The layer that is being edited is random.

Following your workflow does not reproduce that to me.
To me it will select the current layer that is on the preview.

If I use the tool, apply and reopen, it will select the exact same layer I was editing.
I guess your problem is that you navigate up/down into layers after apply the tool, then reopen the tool causes to start on the current displayed layer. If that's not the case then provide a small video showing the issue. Otherwise, this is not an issue but is by design.

Hazoooka, 2023-06-20:
Attachments available until 20 Jul 2023
Here, I attach two videos showing the same issue in 2.29.0 in Mojave and 2.14.0 in Monterey.

By the way, following your tip about the hidden preferences I have copied the whole .local folder from Mojave to Monterey and the preferences now work as intended, I can save the settings.

Yet the problem with functionality of some parts of your program remain.

Kind regards,
Witold.

Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAWgPCUMQ-c5f3XEwwmlLm-d0uUj-AWT3OfFjDnCU5n3RI0qFPN0cBVNu%2F%24%7Bf%7D%3Fo%3DAmg1fGUCcc0MgPy6XomkUnu_c6_Jt1sJEJR8EM7gbQPa%26v%3D1%26x%3D3%26a%3DCAogqWXlwXpSMnzq8j4mpjlQ9F6Bj-pJT4IedPuFeqZTWwMSeBDD48_EjTEYw_PKmJcxIgEAKgkC6AMA_1pJLN9SBHS5SP5aBBwFU25qJnDP8Lgg3Ej_kwcb04iV2XlssoGAriVJPQ94JSsOTNMj4KtmiDdYciadXOuYITwv5aU8Uwmea2VDN3WADI7CU7tr6AIaPZEGceCMZxT_MA%26e%3D1689852754%26fl%3D%26r%3D6D5B79EE-3FC9-476E-9FBD-9A548AE87CFA-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3DFF830BCF-C728-4BB5-86FA-35621CF66201%26p%3D116%26s%3Dn0LuhRzEAha2uzgpMQA1wlh1l88&uk=Wm-pZzIner1_Nqv3wo1AIQ&f=2.29.0%20Mojave.mov&sz=8965026>2.29.0 Mojave.mov
9 MB
Click to Download
 <https://www.icloud.com/attachment/?u=https%3A%2F%2Fcvws.icloud-content.com%2FB%2FAbgDcwfmdBURjwpgFQ9jTonumLv_AYBPinJGa8DsB_bvFnmLxTsTAgVI%2F%24%7Bf%7D%3Fo%3DAnrLJRILc3oluYX70eerzMFrUlhv2hePPiZWGJxlc1NW%26v%3D1%26x%3D3%26a%3DCAog9wqUeL1yMxDa7csl5Qv45Z9tTPWZ5qrANtSgggqOipoSeBDw5M_EjTEY8PTKmJcxIgEAKgkC6AMA_wKQUNdSBO6Yu_9aBBMCBUhqJoy1TpJ87blRBXKBGD9WMKl8MwuJBhiIkSLlZTiJOBHC8U0Xs2dEciZPLJsHwPfW7oRtrBuEb8Sy3IKs0oBYoTvC2BfT4CMtX_4d6EvvjQ%26e%3D1689852754%26fl%3D%26r%3DBA752B7E-53EA-4F00-A9A4-171AAC44AD12-1%26k%3D%24%7Buk%7D%26ckc%3Dcom.apple.largeattachment%26ckz%3DFF830BCF-C728-4BB5-86FA-35621CF66201%26p%3D116%26s%3DfZOnePZKZ6TasGELCHFLBvAtZjE&uk=RfjAkBicN-PliyTrTwt1Xg&f=3.14.0%20Monterey.mov&sz=12958196>3.14.0 Monterey.mov
13 MB

Hazoooka, 2023-06-20:
A tip about UNIX systems.

You do not hide the preferences or settings files, more you store them in god knows where.
There is a standard to this.
You store them in the Applications Support folder, or a Preferences folrder in the Library folder of the account.

Kind regards,
Witold

sn4k3, 2023-06-20:
When you have detected issues after applying any tool, it will refresh data and UI, that cause to select the first issue on the list, making it to display the containing layer. Thats the reason why after you apply the edit parameters it goes to layer 0. That is not a bug, is just the way UVtools currently works by showing the first issue after a refresh.
This should not be a problem when I workout your request for not closing the window

Hazoooka, 2023-06-20:
The issue detection is turned off in the settings in both 2.290 and 3.14.0, so this is not the case.
Besides you can see no issues tab turned on.

The program sometimes jumps to the top layer instead of 0, and sometimes it’s following the selection.
The problem is how the program is organised in this area - it’s not organised, it chaotic from what I can see.

sn4k3, 2023-06-20:
Clearly you have an empty layer at layer 0, both videos show that, how do you tell me you don't!.

[image]

> The issue detection is turned off

Doesn't matter, empty layers and print height always detect if they are enabled on settings (Issues - Compute). 
"Auto compute issues on file load" is for the time expensive detections where make sense to opt.

> Besides you can see no issues tab turned on.

Doesn't matter if tab is selected or not, as long you have issues detected that's the default behavior after apply any tool.

> The program sometimes jumps to the top layer instead of 0

Because you have an issue on last layer, most likely another empty layer as some slicers do that.

> and sometimes it’s following the selection.

Because you have no visible issues on file

Ignore or remove all issues from list and you will see that the selection is followed with the current selected layer

Hazoooka, 2023-06-20:
So you are saying that even if I turn everything off from the issue detection panel it’s detecting anyway?

Look, this is my issue detection settings:

Kind regards,
Witold.

Hazoooka, 2023-06-20:
I can not remove the issue, I can only remove the problematic layer, which I can not do because it’s Layer 0, it has to be empty.
Ignoring the issues works, the program is following the selection, thank you.

Can you make the program to ignore the issues by default when the detection is turned off? Or add a separate checkbox to the settings to make  them ignored by the program?
So many tiny spoilers must be remembered to use UVtools that it becomes almost like using windows applications, constant problems.
Users needs to focus on doing their jobs, not to remember all the little tweaks before even starting their work.
All should be in the settings, set once.

Have you read my suggestion about editing and saving all the layers first and then closing the edit window?
That would be much more convenient.

Have you ever been thoroughly testing the program to find the inconsistencies and issue?
I would be happy to help testing it so it is more user friendly.

Kind regards,
Witold.

> On 20 Jun 2023, at 18:19, Tiago Conceição ***@***.***> wrote:
> 
> Ignore or remove all issues from list and you will see that the selection is followed with the current selected layer

sn4k3, 2023-06-20:
> So you are saying that even if I turn everything off from the issue detection panel it’s detecting anyway?

Yes, instant to calculate issues (PrintHeight and EmptyLayer) will always show (If present) on file load no matter the settings, because they cost no time to detect.

> Look, this is my issue detection settings:

Image is missing, attachments via email replies does not work well with github 

>  I can only remove the problematic layer, which I can not do because it’s Layer 0, it has to be empty.

Why layer 0 must be empty in your case?

> Can you make the program to ignore the issues by default when the detection is turned off? Or add a separate checkbox to the settings to make  them ignored by the program?

I can add an extra setting to improve the current situation where user can fully disable any detection on file load.

> So many tiny spoilers must be remembered to use UVtools that it becomes almost like using windows applications, constant problems.

You keep using that relation with Windows when it's not related. macOS apps are not a synonym of perfection nor bug/problem free nor beauty nor well-made or whatever you want to list. Nothing prevents you from doing an excelente or bad app, that is valid for every OS and systems.
What Apple fan-boys don't realize is that Windows is made to work in thousands of different combinations of hardware and specs, and then is a hard task to maintain such combability, this already makes the Windows a big feat and a beast, and so have it pros and cons, still for what it is, is very well optimized and well made, what bring it down it's the bad developers and bad driver makers, most of the problems users have comes from 3rd party drivers, apps and the user itself, not windows.
In the otherwise, Apple make macOS just to a small number of devices that you can count by your hand, so the system is optimized for that selected hardware and so they have much easier task in maintaining the OS fluid and with great performance, because it's a controlled eco-system, that comes with the cost of limited hardware chooses, a bind to apple products and a limited time for system upgrades as they drop support for older macOS as new releases comes out (This is again because it's easier to mantain the system and also make a better one where you don't need to have combability with past hardware).
IMO user should use what it likes most, feels comfortable or what is best for them, some tasks may be better on one than another, you can have a macOS on home and a Windows on your job and still love both for what you do.
The fan-boys war Windows vs macOS for me is non-sense, both are good and both are bad, it depends on your task, needs, habituation, etc. If you most like macOS buy apple, if you like Windows buy a PC is easy as that and none is the mighty god of the OSes.

Back to UVtools, its complexity and number of features certainly will not please everyone, that's normal when you have to deal with big programs that are bloated with options and functions. You have to deal with that, and as everything else in life if you want something exactly at your taste, the sometimes the only option is to build it yourself. You may like UVtools to behave in one manner, person B may like in another manner different from you, and so on, even 1000 persons may all be in disagreement. That's the reason why exists many programs that do the same propouse, but with minor alterations in function and settings, ie: video player, foto viewer, ect. Unfortonally in this category there's only the UVtools and so people are locked to one software that is build incremental and adapted as it grows. So, it's hard to mantain such program in a free manner and even still give pristine support.

> Users needs to focus on doing their jobs, not to remember all the little tweaks before even starting their work.
All should be in the settings, set once.

I agree, however sometimes is not possible to have all, we just need to adapt and live with it. The more you use the program that annoyances will ease. Also, we are in an era where people want everything as they want and one-click solution without any effort, people are becoming lazier and that's a problem. 

> Have you read my suggestion about editing and saving all the layers first and then closing the edit window?
> That would be much more convenient.

Yes I did, in case you miss it I mentioned that [here](https://github.com/sn4k3/UVtools/issues/730#issuecomment-1599043448), I will see what I can come with to implement that.

> Have you ever been thoroughly testing the program to find the inconsistencies and issue?
I would be happy to help testing it so it is more user friendly.

There is a large number of users using the program, and so they are all testers. When they find issues, they come here to report them. Program evolve based on users reports and suggestions, still not easy to mantain such large program, touching critical aspects can break stuff, and changing others can be such a hard task that for me does not make sense (Again this is a hobby without income) Even still people are getting premium support and fast patches.

Hazoooka, 2023-06-20:
> we just need to adapt and live with it

And that kind of thinking is the main blocker of improvements.

> touching critical aspects can break stuff

I think it’s a matter of preparation and method that is used. In every job proper/improper preparation/method can make a huge difference.

Step by step, you can manage to make this app better and better.
It will take some work of course, but you chose this path, it was obvious it will require a lot of work.

I’m wondering about the users testing your app.
What I am pointing out is the first thing to correct for a reasonable user. It’s a matter of user experience and workflow efficiency.
I bet users were using editing layers a lot, and yet so many releases and nothing! No one reported this!
Or perhaps they are too polite, or they simply do not care? Perhaps they adjust?

As for the applications you are right, it’s up to the programmer who knows what he’s doing or hasn’t got a clue! And as a result we have workable applications or the ones one does not know what to start with.
System of course makes a huge difference as well.
Having messed up architecture it’s much more likely to end up with a messed up program.
Having organised architecture is much of success.
Simple as that.

I’m looking forward to the updates.

Kind regards,
Witold.

Hazoooka, 2023-06-20:
Because of the elephant’s foot.
No the safe way but to fiddle with wait before print.
For this to work first layer MUST be empty.
Unfortunate feature of the printer’s firmware to skip the first layer’s parameters.
The only way to overcome this is to make the first layer empty.

> On 20 Jun 2023, at 20:24, Tiago Conceição ***@***.***> wrote:
> 
> Why layer 0 must be empty in your case?

Hazoooka, 2023-06-20:
But you already know this.
You are working closely with Lychee team.

Kind regards,
Witold.

> On 20 Jun 2023, at 22:37, Witold Hazuka ***@***.***> wrote:
> 
> Because of the elephant’s foot.
> No the safe way but to fiddle with wait before print.
> For this to work first layer MUST be empty.
> Unfortunate feature of the printer’s firmware to skip the first layer’s parameters.
> The only way to overcome this is to make the first layer empty.
> 
>> On 20 Jun 2023, at 20:24, Tiago Conceição ***@***.*** ***@***.***>> wrote:
>> 
>> Why layer 0 must be empty in your case?
>

Hazoooka, 2023-06-20:
Hi Tiago.

Would you have some time to explain the suggestions panels settings to me, some of them?
It’s really a pressing matter for me as I’m trying to use those to harness the elephant’s foot and those settings seem to be the right way.

Kind regards,
Witold.

sn4k3, 2023-06-20:
> Because of the elephant’s foot.
No the safe way but to fiddle with wait before print.
For this to work first layer MUST be empty.

Remove all empty layers with UVtools and then apply the Wait time before cure suggestion. It will create an "empty layer" but with a single faded pixel which solve the annoyance of empty layer detection.

> Would you have some time to explain the suggestions panels settings to me, some of them?
It’s really a pressing matter for me as I’m trying to use those to harness the elephant’s foot and those settings seem to be the right way.

I can answer in the other tropic you created about that. You can trust UVtools and apply the defaults, they will never do harm and uses a time/quality profile.
If you right-click on suggestion and click on "More information (Web)" it will lead to jan website with detailed problem of EF. If you follow the 3 articles you will find the last about UVtools and explaining some of parameters.
https://blog.honzamrazek.cz/2022/01/prints-not-sticking-to-the-build-plate-layer-separation-rough-surface-on-a-resin-printer-resin-viscosity-the-common-denominator/
https://blog.honzamrazek.cz/2022/07/step-by-step-guide-on-perfect-bed-adhesion-and-elephant-foot-removal-in-uvtools-3/

Hazoooka, 2023-06-20:
> It will create an "empty layer" but with a single faded pixel which solve the annoyance of empty layer detection.

This one pixel is messing up the whole idea.
The idea is to have NO pixels, nothing is cured and giving the thickness.

> You can trust UVtools and apply the defaults, they will never do harm and uses a time/quality profile.

It’s also not giving me what I need in default settings, that’s the problem.

> If you right-click on suggestion and click on "More information (Web)" it will lead to jan website with detailed proble

I did, there is nothing explaining this particular section.

I know Honza Mrazek’s articles.
He explains the same way: "do it by default, for me it works” No sense in this.
I need a step by stem manual of each option to be able to use it consciously, not blindly hitting “yes” to everything.

That’s one thing missing big time, a proper manual explaining the functions in detail.

Kind regards,
Witold.

sn4k3, 2023-06-20:
> I bet users were using editing layers a lot

Not really, global parameters yes, but per layer editing no. People who really want to edit layers parameters create scripts to automate their workflow, that way they reuse the script and just call it one time to do the job. UVtools scripting allow you to make any change on file in a programmatic way.

> This one pixel is messing up the whole idea.
The idea is to have NO pixels, nothing is cured and giving the thickness.

It's not a problem, nothing is cured. UVtools create a pixel very faded which cure for about 0.01s. Even if you cure that single pixel for 1s it does not form a pixel because brightness is low and light will have a hard time passing. 

> It’s also not giving me what I need in default settings, that’s the problem.

If you still have EF you need to increase the wait time. You have to test and increment, it's the only way to be on spot. As told default values are conservative to not impact time too much and at same time try to get the most of EF off. If not optimal workout the defaults and increase "wait time". The bigger the screen the more time you need, the thicker the resin the more time you need, less rigid Z the more time you need, etc, etc, there are many factors that only by experimentation you will obtain the optimal value.

> I need a step by stem manual of each option to be able to use it consciously, not blindly hitting “yes” to everything.
> That’s one thing missing big time, a proper manual explaining the functions in detail.

I have no plans in make detailed manual. However, I can answer your questions but in the other thread you created, just hold a bit.

Hazoooka, 2023-06-20:
> On 21 Jun 2023, at 00:28, Tiago Conceição ***@***.***> wrote:
> 
> If you still have EF you need to increase the wait time. You have to test and increment, it's the only way to be on spot. As told default values are conservative to not impact time too much and at same time try to get the most of EF off. If not optimal workout the defaults and increase "wait time". The bigger the screen the more time you need, the thicker the resin the more time you need, less rigid Z the more time you need, etc, etc, there are many factors that only by experimentation you will obtain the optimal value.

I know all this.
I need to learn the Suggestions settings to harness automatic wait time assignment.
I do not know how to make the script and suggestions allows me to do what need.
The only problem is that there are some settings I do not know what for, what they are referring to

Witold.

Hazoooka, 2023-06-20:
Regarding the scripts, Honza wrote a few scripts for UVtools, for warping and some other issues, but none of them are working!.
Every time I get the “void ScriptInit()” message.
So much for scripting.

Witold.

> On 21 Jun 2023, at 00:37, Witold Hazuka ***@***.***> wrote:
> 
> 
> 
>> On 21 Jun 2023, at 00:28, Tiago Conceição ***@***.*** ***@***.***>> wrote:
>> 
>> If you still have EF you need to increase the wait time. You have to test and increment, it's the only way to be on spot. As told default values are conservative to not impact time too much and at same time try to get the most of EF off. If not optimal workout the defaults and increase "wait time". The bigger the screen the more time you need, the thicker the resin the more time you need, less rigid Z the more time you need, etc, etc, there are many factors that only by experimentation you will obtain the optimal value.
> 
> 
> I know all this.
> I need to learn the Suggestions settings to harness automatic wait time assignment.
> I do not know how to make the script and suggestions allows me to do what need.
> The only problem is that there are some settings I do not know what for, what they are referring to
> 
> Witold.

sn4k3, 2023-06-20:
> Every time I get the “void ScriptInit()” message.

Either you are using a old version of UVtools or script have errors that need fixes

Hazoooka, 2023-06-20:
3.14.0 is not an old one

sn4k3, 2023-06-20:
Then the script have errors that need fix. Error message most of the time tell what is the problem.

sn4k3, 2023-06-21:
You can now opt to not detect issues.

closed 2023-06-21 by sn4k3

Hazoooka, 2023-06-21:
Thank you

Kind regards,
Witold