sn4k3/UVtools #731: [Bug]

Hazoooka, 2023-06-19:
### System

```Text
UVtools v3.14.0 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Mon Apr 24 21:10:53 PDT 2023; root:xnu-8020.240.18.701.5~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
Processor cores: 24
Memory RAM: 9.16 / 24.00 GB
Runtime: osx.12-x64
Framework: .NET 6.0.16
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Users/witos/Applications/UVtools.app/Contents/MacOS/
Executable: /Users/witos/Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: cube.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Description of the bug

Edit print parameters window is closing every time we save new parameters.

Please make it stay opened until we finish editing all the layers we need.

### How to reproduce

Open the "Edit print parameters window"
Make some change and save it.
observe how inconvenient it is to open this window again and again, every time one needs to edit something.

sn4k3, 2023-06-21:
Implemented

closed 2023-06-21 by sn4k3

Hazoooka, 2023-06-21:
That was fast, thank you!

Witold.