sn4k3/UVtools #738: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Object reference not set to an instance of an object.)

BenjaminSchuermann, 2023-07-04:
### System

```Text
UVtools v3.14.4 X64
Operative system: Linux 6.3.5-2-MANJARO #1 SMP PREEMPT_DYNAMIC Sun Jun  4 18:26:12 UTC 2023 X64
Processor: 12th Gen Intel(R) Core(TM) i7-12850HX
Processor cores: 24
Memory RAM: 14.19 / 62.41 GB
Runtime: manjaro-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /home/garog/Downloads/UVtools_linux-x64_v3.14.4/
Executable: /home/garog/Downloads/UVtools_linux-x64_v3.14.4/UVtools
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Object reference not set to an instance of an object.)
 ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.LayerCache.set_Layer(Layer value) in UVtools.WPF/LayerCache.cs:line 40
   at UVtools.WPF.MainWindow.ShowLayer() in UVtools.WPF/MainWindow.LayerPreview.cs:line 1499
   at UVtools.WPF.MainWindow.ComputeIssues(IssuesDetectionConfiguration config) in UVtools.WPF/MainWindow.Issues.cs:line 688
   at UVtools.WPF.MainWindow.OnClickDetectIssues() in UVtools.WPF/MainWindow.Issues.cs:line 631
   at UVtools.WPF.MainWindow.<set_SelectedTabItem>b__49_0() in UVtools.WPF/MainWindow.axaml.cs:line 207
   --- End of inner exception stack trace ---
Category: Task
File: termi_faction_legs_3h42m.ctb  [Version: 0] [Class: CTBEncryptedFile]

Machine date time: 04.07.2023 14:00:13
    UTC date time: 04.07.2023 12:00:13
```

### How to reproduce

Have a .ctb file open and then select open to open a new ctb file

sn4k3, 2023-07-04:
Duplicate of https://github.com/sn4k3/UVtools/issues/648
I cannot reproduce this. Can you?

BenjaminSchuermann, 2023-07-05:
no, but i got that crash a few times, i will take care of it if i do something special to reproduce it

sn4k3, 2023-09-17:
Maybe fixed in last version, reopen if needed

closed 2023-09-17 by sn4k3