sn4k3/UVtools #739: Importing a PNG image results in messing up Lychee pre-saved variable-parameters of the layers

Hazoooka, 2023-07-14:
### System

```Text
UVtools v3.15.1 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
Processor cores: 24
Memory RAM: 10.74 / 24.08 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Volumes/Macintosh HD/Users/witos/Applications/UVtools.app/Contents/MacOS/
Executable: /Volumes/Macintosh HD/Users/witos/Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: BMV-803 case.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Slicer: Lychee

### Description of the bug

Variable layer parameters exported from Lychee are messed up by the import layers action.
When I save a CTB with different layer thickness, 'exposure' and 'wait before print' times it loads into UVtools with proper parameters.
When I replace a layer all previously set up parameters go back to some default value.
All layer thicknesses ('position Z'), 'exposure time' and 'wait before print time' are set to a single value, erasing the ones exported from Lychee.
It happens every single time a layer is imported, no matter which import option is selected: 'replace", 'stack', 'merge' or any other, the import erases the parameters.

Checked in 3.14 and 3.15 in Catalina, as well as 2.29 in Mojave.
The same result.

### How to reproduce

- make a multi-parameter file
- check the parameters of the layers in the files
- open it in UVtools
- select the 'import layer' action
- select select whatever layer number you want to replace
- choose the file and import it
- check the layers parameters again and observe how everything is changed

The layer number selected for import makes no difference.
The name of the file makes no difference, 'layer000.png' triggers the same bug as any other .png name.

sn4k3, 2023-07-15:
Will be fixed on the next patch

Hazoooka, 2023-07-22:
Please fix this bug also in 2.29.0
Some users are still using Mojave due to the hardware limitations for newer systems.

Regards,
Witold.

Hazoooka, 2023-07-25:
I have just checked that this bug is also affecting usual, single parameter prints.
Importing even one bitmap into a print previously edited in UVtools brings the settings to the original state erasing all edited changes.

Kind regards,
Witold Hazuka.

sn4k3, 2023-07-25:
> this bug is also affecting usual, single parameter prints.

Because it resets the layer properties no matter what.
Until patch there's no workaround other than via scripting...

> Please fix this bug also in 2.29.0

Only the current branch gets updates. All past versions are frozen as it is.

sn4k3, 2023-09-17:
Fixed

closed 2023-09-17 by sn4k3

Hazoooka, 2023-09-20:
Thank you, good job.

Kind regards,
Witold

Hazoooka, 2023-10-18:
Hello Tiago.

I have been using your UVtools a bit more extensively now, and I must admit it’s a really powerful tool.
Apart from several issues that surprised me in the beginning I must say that you really know your stuff.

Well done.

Kind regards,
Witold.