sn4k3/UVtools #741: Edit print parameters does not allow changing the layer's thickness

Hazoooka, 2023-07-16:
### System

```Text
UVtools v3.15.1 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64 X64
Processor: Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
Processor cores: 24
Memory RAM: 9.73 / 23.89 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Users/witos/Applications/UVtools.app/Contents/MacOS/
Executable: /Users/witos/Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Hull details.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Description of the bug

When editing the print parameters one can change many parameters but the layer thickness.
There is 'Position Z' parameter actually pretty useless, changing 1000 layers thickness by giving each a new height one by one is madness.
Instead it should be possible by simply selecting the layers range and changing their thicknesses, just like all other parameters.

There is a 'Dynamic layer height' function but greyed out, not accessible, so no idea what it can or can not do.

### How to reproduce

Open the 'Edit print parameters' > 'Change settings per layer range' and see the layer thickness missing.

sn4k3, 2023-07-16:
"Edit print parameters" works with modifiers that impact global properties or layer properties. It can't do what you want.
Instead look at this tool:

[image]

> There is a 'Dynamic layer height' function but greyed out, not accessible, so no idea what it can or cannot do.

greyed-out mean can't be used by your file/printer. But its purpose is to make dynamic slices.

Hazoooka, 2023-07-16:
Ah, right, missed the drop down menu with the layer range.
Thank you!

Hazoooka, 2023-07-17:
‘Adjust layer height’ by the layer offset is not working as intended. It’s not changing anything in the layers thicknesses, no matter if I give positive or negative values, the thicknesses remain and the overall model height remain the same.

sn4k3, 2023-07-17:
Offset like the name says, offset all layers from the starting point, it does not affect thickness of each. It common use is to offset the starting height of the print (First layer) without affect anything else.
You want the method: "Re-height" which will change all layers height and redo all layers, including the properties.

closed 2023-07-17 by sn4k3

Hazoooka, 2023-07-17:
The offset as the word describes and the menu of the function suggests, offsets the range of layers thicknesses by a specified value.
As a result I’m expecting each layer to be offset by this value.

The description of the function says:
"Adjust the layer height of the model or offset layers position by a set value.
Adjusting to values lower than current height will reduce layer lines, adjusting to values higher than current height will reduce model detail.
Different layer thickness will require different exposure times, adjust accordingly”

This exactly means that I am changing the thickness of the chosen layers.

Please put this thing in order.

sn4k3, 2023-07-17:
The description is totally correct. Read it again, specially the first line: 

"Adjust the layer height of the model or offset layers position by a set value."
.......................................................................... **OR** ..........................................................................
-----------------------------------------^

So, the OR is there for a reason? It does one of the two functions which are different, not both.

Then you choose the method which tell you what will be performed:

[image]

Layer height/thickness is not the same as layer position, they have indirect correlation but definitely they are not the same thing.
From my observations looks like you almost use the tools randomly/blind without thinking in what you are about to use, descriptions are there, tools are there you just need to careful read and think about the stuff you are end to use.

For example, your sentence is proof that you don't read any of UVtools labels for real.
> The offset as the word describes and the menu of the function suggests, offsets the range of layers thicknesses by a specified value.

You said that the menu of UVtools suggests: "Offsets the range of layers thicknesses by a specified value" when that is not true! Menu item says: **Offset - Change layers position by an offset** and again **layer position** is not **layer height/thickness**.

To be clear I can explain what those are:
**Layer height:** The thickness of individual layer in mm (Configured in slicer)
**Layer position:** The absolute Z position in mm where layer will be print

You were using Offset, now, please read this again: **Offset - Change layers position by an offset**. 
I think you will now agree that method is telling that will Offset ALL selected layer range **positions** by an offset. If so, it means ALL positions will be offset, and it will "not" affect layer height, why? Well because if you offset everything equally the difference is null/same.

See this example:

```
Layer 0 pos: 0.05mm
Layer 1 pos: 0.10mm
Layer 2 pos: 0.15mm

Layer height = Current layer pos - previous layer pos.

layer 0 height = 0.05 - 0mm = 0.05mm
layer 1 height = 0.10 - 0.05mm = 0.05mm
layer 2 height = 0.15 - 0.10mm = 0.05mm
```

Now let's offset them all by 1mm!

```
Layer 0 pos: 1.05mm
Layer 1 pos: 1.10mm
Layer 2 pos: 1.15mm

layer 0 height = 1.05 - 0mm = 1.05mm
layer 1 height = 1.10 - 1.05mm = 0.05mm
layer 2 height = 1.15 - 1.10mm = 0.05mm
```

As you can see offset layers will only impact the starting layer height in the range AND the next layer after the last on the rage (if any), all other subsequent layers are unaffected in thickness because you offset them all equally.
This is the reason why I said: " Its common use is to offset the starting height of the print (First layer) without affect anything else." so you can modify the starting position of the print closer or far from the LCD. Of course, there are other uses for the offset method but it doesn't matter for the case.

Now read method 1: "Re-height: Change layer height / thickness", it won't let you to choose a range because it will impact all layers and change every layer thickness. It also presents you another menu to choose a modifier of the current height (Yes, you can't choose any value) this is to not lose any quality, the difference must be in multiples. This will re-do all layers without affect print height, so no matter the height changes the final model height will be the same (Not become stretched), hope that makes sense...

___

Don't take what I will tell next as an insult or depreciation but rather a hint and advice to improve and be better:

I'm professor on my local university and the pattern these days are very bad, students (+18 years) are entering in advanced lectures with very bad logical and mentally health, to see how bad it is, one day I asked to do a function that returns the cubic number of its argument (Math cube), and I was amazed to almost see "smoke" coming out of their head. How such simple operation could be soo hard for them as that is given in elementary school??? I told: forget programming! How do you do it on paper like in math, for example, calculate the cube value of 4? Once again, they were unable to solve!!! To see this on my first year was very deceiving giving the "advanced" engineering course in electronics. I once tried to do exams with full open consulting, and god-damit they fail miserably, even with copying from my sources most managed to mix everything, misspell words etc. In other words, even with the solution on hand they managed to ruin it because they are lacking on a fundamental factor (Brain!).
All subsequent years were worse, I notice that year by year they come totally unprepared and with a huge lack of logical thinking and mental agility due to tech and other brain wash events. PS: Brain hates virtual stuff and will under-develop if using tech instead of physical objects (Thats a problem for new generations that born with a tablet/console/smartphone) in their hands and we are starting to see that in first hand.
Another fun event: One student lost 30 minutes of his Linux exam trying to find the "{" key on his mac keyboard to complete the command 🤦I had to go there and do it for him and say the combination stroke. And what was the problem? Keyboard don't have the symbol printed in key and then he was unable to try every key combination on the number/symbol's rows, worse, he unable to jump to the next question! [... cut ...] ove but also be better in much other aspects, including outside of this program scope.

I hope this text can help you going in the correct direction and you can take something good out of it. 
Realize that most people would just tell you to **** off in your back and don't care at all about you and end comms with an "ok" or something cheap/fake polite. Instead, I'm here "wasting" time that I can never recover in a possible hope that I can help someone positively. So, the following is up to you...

Hazoooka, 2023-07-17:
I’m looking for a workaround to the bug which deletes the layer’s parameters upon importing the bitmap slice images.

> "Edit print parameters" works with modifiers that impact global properties or layer properties. It can't do what you want.
Not true, it can also modify each layer separately, locally.
I see no problem with changing layer’s thickness among all other parameters.

Hazoooka, 2023-07-17:
> Realize that most people would just tell you to **** off in your back and don't care at all about you and end comms with an "ok" or something cheap/fake polite.

Similarly some people tells you exactly the same to some functions you make without giving them any practical thought, without testing them properly, making eventually your program useless to some users.