sn4k3/UVtools #742: [Bug] UVTools produces corrupted pm5s files for the Anycubic Photon Mono 5s

cdheiser, 2023-07-19:
### System

```Text
UVtools v3.15.1 X64
Operative system: Darwin 22.5.0 Darwin Kernel Version 22.5.0: Thu Jun  8 22:22:22 PDT 2023; root:xnu-8796.121.3~7/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz
Processor cores: 16
Memory RAM: 15.84 / 28.12 GB
Runtime: osx.13-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Sreens, resolution, working area, usable area:
1: 1792 x 1120 @ 200% (Primary)
    WA: 1792 x 1095    UA: 896 x 548
2: 3200 x 1800 @ 200%
    WA: 3200 x 1775    UA: 1600 x 888

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Diablo One.pm5s [Version: 518] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer:  Anycubic Photon Mono M5s
- Slicer:  Prusa Slicer 2.6.0

### Description of the bug

Initial generated pm5s files are corrupted resulting in a print that does not proceed past the first layer.  Print times displayed are several orders of magnitude greater than what's expected (hundreds of hours), and never proceeds past the first layer.

I've made both the bad and working pm5s files available here: https://file.io/JoWgslx6cLfQ  as well as the sl1 file these were converted from.  I can also share the original prusa project.

### How to reproduce

1. Slice a file in Prusa Slicer
2. Export in SL1 Format
3. Open in UVTools
4. Copy converted pm5s file to the printer
5. Print the file
6. Observe the print time looks correct until the first layer goes to print and then jumps by a factor of approximately 100x

If you reopen the pm5s file and make a no-op change and re-save it, the resulting file prints successfully.

sn4k3, 2023-07-19:
If I understand correctly, the conversion out of SL1 generates a bad file, but if you reopen the file and force a resave without any other change it makes files good?

cdheiser, 2023-07-19:
That's correct.  I've spent the last few days figuring out how to
consistently reproduce the problem.  My method of forcing a re-save is to
edit the print settings to increase the base layers by 1 and then set it
back.

cdheiser, 2023-07-19:
I forgot to mention that this happens on both Windows as well as MacOs

sn4k3, 2023-07-19:
There is a better way to force a full save: Pixel editor and erase a black pixel on a black pixel (Does nothing but turns the full encode flag). Also, it's possible to do that via Terminal with two lines of code.

What I find between both:
Model XY info is missing on bad file, but I think it's not required, anyway [I provide a file](https://we.tl/t-fY0f1vEUOO) with only that fixed to you to try and report back.
Then the per layer settings is enabled which will follow layer table instead of the global settings, while your good file have that set to false and I confirm that force a full encode without any other setting turn that to false

cdheiser, 2023-07-19:
The file you provided had the same behavior.  I've attached a photo of the status screen on the printer.
[image]

I ran this test with the build plate and vat off so I could see what's going on.  The first layer does what you'd expect, 10 second exposure, but then it just sits there, screen off, like there's some extra wait time it needs to do.

cdheiser, 2023-07-19:
> Can you check if theres an firmware upgrade for your printer?
> Because that is not normal, it is trip by follow layer table and those are
> correct set as I have checked.
> Please report back with your firmware version

The printer is on 1.0.2.1 which is the latest version on Anycubic's site.

sn4k3, 2023-07-19:
Ok, so let's do another test... 
Slice a file on Anycubic slicer, test if its printing and attach here. (Without UVtools processing)

cdheiser, 2023-07-19:
I sliced this file in Anycubic Photon Workshop and it prints without
error:  https://file.io/PEquEQfzs8y3

sn4k3, 2023-07-19:
Now try the same file but with per layer flag on (Manually edited): [Diablo WingsPerLayer.zip](https://github.com/sn4k3/UVtools/files/12099704/Diablo.WingsPerLayer.zip)

However, that produced file uses version 518 and UVtools was using 517.

sn4k3, 2023-07-19:
Then is a firmware bug, and a bad one that prevent you from use that feature. 
UVtools have a bug in that conversion stage that set flag ON when it shouldn't, will be fixed on next release.

For now, use this workaround and save:

[image]

sn4k3, 2023-07-19:
Then is a firmware bug, and a bad one that prevent you from use that feature. 
UVtools have a bug in that conversion stage that set flag ON when it shouldn't, will be fixed on next release.

For now use this workaround and save (Reset layers properties):

[image]

DanJaramillo, 2023-08-23:
I have the same issue when running Lychee Slicer 5.2.201 to slice followed by UV Tools 3.15.1 to post process. The Lychee generated file displays correct times and layers, however the UV Tools file does not print beyond the first layer, show a correct preview, or show correct times. This is evident on both the AnyCubic M5S interface as well as the AnyCubic Photon Workshop.

sn4k3, 2023-09-17:
Fixed

closed 2023-09-17 by sn4k3

Rubiniuz, 2023-09-28:
So the above comment

> Then is a firmware bug, and a bad one that prevent you from use that feature. UVtools have a bug in that conversion stage that set flag ON when it shouldn't, will be fixed on next release.
> 
> For now use this workaround after conversion and save again (Reset layers properties):
> 
> [image]

this is the fix? cant this be fixed in a uv tools update? or is this a anycubic issue?

just clicking that button fixes the issue? or are there more steps required?

I really want to use uv tools but i also get that first layer issue. a step by step guide also posted on reddit could save a lot of people a lot of time

sn4k3, 2023-09-28:
It's fixed on v4, however the bug persist on the firmware, if you use the suggestions on UVtools it will enable the flag that trigger the problem. If you only use to fix issues it won't be enabled.

If and only after you save the file the following highlight "**PerLayerOverride**" is **True** printer will produce that problem

[image]

In your case if you see True you must turn that False by the trick of "reset layer properties".

Skorinn, 2023-10-23:
I am able to reproduce this bug with 4.0.3:

[image]

After slicing the file with Lychee, and openning in UVTools the PerLayerOverride property is False:

[image]

I used the Pixel Editor to delete a blank pixel and saved the file (Save As). After which the PerLayerOverride is True:

[image]

Using the Reset Layer Properties does reset it back to False. Please let me know if you need any additional information.

Thanks

Rubiniuz, 2023-10-23:
> I am able to reproduce this bug with 4.0.3:
>
> [image]
>
> After slicing the file with Lychee, and openning in UVTools the PerLayerOverride property is False:
>
> [image]
>
> I used the Pixel Editor to delete [...]

Yeah exactly. thats why we closed it. thanks for the heads up again

Skorinn, 2023-10-24:
> Yeah exactly. thats why we closed it. thanks for the heads up again

I'm not sure I follow. Based on the comments above, I thought that 4.0.0 included a fix for UVTools to not always set PerLayerOverride to True when using it to detect and repair issues (i.e., islands, resin traps, etc.). Is that not the case?

If not, I'm curious if there are plans to make that change or if this is the intended behavior? The workaround provided (thank you) is easy enough to make this a low priority, but it is easy to forget to do,

If the change has been made, is there something I am doing incorrectly that is resulting in that property getting set to True? What triggers that change?

Thanks

sn4k3, 2023-10-24:
You need to provide the file, for me replicate.

> I thought that 4.0.0 included a fix for UVTools to not always set

Yes, until you apply suggestions or have them to auto apply the `PerLayerOverride` will turn true.
Also if slicer sets a individual value on layer different from global it will cause UVtools to set that to true.

Skorinn, 2023-10-24:
> You need to provide the file, for me replicate and find the problem.

Here is the file created by Lychee before post-processing (PerLayerOverride=False):  [M5_100um_3DSLA_TestModel.pm5s](https://1drv.ms/u/s!AvtXD5C8KSR9iPsREVxeEkVI3jOtkg?e=3EbWqh)

This is the file saved by UVTools after solely using the Pixel Editor to delete a black pixel (PerLayerOverride=True): [M5_100um_3DSLA_TestModel_00h32m_1ml_copy.pm5s](https://1drv.ms/u/s!AvtXD5C8KSR9iPsVrAbNA6kPYEoGFA?e=08jcQc)

Thanks

Skorinn, 2023-10-25:
> The problem is on slicer or how you have configured it:

That makes sense. I'll take a look at fixing the profile in the slicer. Thank you for looking into this for me.