sn4k3/UVtools #745: [FEATURE REQUEST] Add dithering capability to improve surface rendering on printed objetcs

AndyVincentUK, 2023-08-11:
### Is your feature request related to a problem?

The Saturn 3 Ultra and some other printers now use a frosted ACF film. Having experimented with this, it works really well on curved surfaces, but really poorly on vertical ones.

The frosted film essentially introduces a level of noise into the printing but randomly modulating the light value. However, this is only random across a layer - the same pattern is applied to every layer since this is a fixed film. This leads to visible lines on vertical surfaces

### Describe the solution you'd like

Rather than implement dithering in hardware (the film below the vat), it would be better if dithering was applied in software with a different set of (random) values applied to each layer. This would ensure that the dither value would be different for the same pixel on successive layers and avoid the limitation with the hardware solution.

I envisage that a random number would be applied to each non-zero pixel. The range could be adjustable - perhaps 10% would apply a random number between 90 and 100% to non-zero pixels - to allow for different printer gray scale capabilities

AndyVincentUK, 2023-08-11:
> Have you tried the Pixel Arithmetic tool? Especially with the "Fuzzy Skin" preset and adjusting the values?
> I think that can do what you aiming for

Hi Tiago

I will try that and report back - it may be a couple of weeks (although I will try later this evening if I get time to change the film) as I go on vacation tomorrow

Many thanks

Andy

AndyVincentUK, 2023-09-09:
Hi Tiago

This works really well (so the enhancement request can be closed). However, is there any more information on how to use the variables? Not all of them seem to have tooltips.

FWIW, my assumptions are (based on Model Surface & Inset):

Wall thickness - presumably the number of pixels below the visible surface that are affected but my testing of this caused me to have doubts about my interpretation!
Ignore areas - not sure how this one works especially as the default is 5000 px squared
Noise range - presumably the upper and lower bounds of the random number added to the pixel value
Noise pixel area - I assume this relates to evaluating more than one pixel but not really sure
Number of passes - presumably this causes the process to run multiple times for the pixel area

I have tried various values and studied the zoomed results but any more accurate guidance would be helpful as this looks like it delivers really great improvements based on my test prints

Many thanks

Andy

sn4k3, 2023-09-15:
Wall thickness: Correct.
Ignore areas: All isolated areas below that threshold will be ignored, eg: To ignore small spots like supports.
Noise range: Correct
Noise pixel area: Applies new random to that chunk of pixels/area. It group pixels and reuse same noise per that area.
Number of passes: Correct

> I have tried various values and studied the zoomed results but any more accurate guidance would be helpful as this looks like it delivers really great improvements based on my test prints

Can't help here, I never used this feature, you need to try and iterate. Also, you can read the [implementation](https://github.com/sn4k3/UVtools/pull/307).

closed 2023-10-27 by sn4k3