sn4k3/UVtools #755: [Crash] Emgu.CV.Util.CvException: OpenCV: could not find a writer for the specified extension

ZJulienZ, 2023-09-11:
### System

```Text
UVtools v3.15.1 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Thu Sep 29 20:12:57 PDT 2022; root:xnu-8020.240.7~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz
Processor cores: 16
Memory RAM: 11.94 / 28.31 GB
Runtime: osx.12-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
```

### Printer and Slicer

- Printer: Phrozen Sonic Mini 8K
- Slicer: 

### Description of the bug

```
I tried to open and save a png file that wouldn't insert using actionbs->import layers

Emgu.CV.Util.CvException: OpenCV: could not find a writer for the specified extension
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
Category: Non-UI
File: VAT - 84 x84 mm _ UVTools_Grids Phrozen Aqua 8K.png  [Version: 0] [Class: ImageFile]

Machine date time: 11/09/2023 15:01:56
    UTC date time: 11/09/2023 13:01:56
```
[VAT - 84 x84 mm _ Grids Phrozen Aqua 8K.png.zip](https://github.com/sn4k3/UVtools/files/12576387/VAT.-.84.x84.mm._.Grids.Phrozen.Aqua.8K.png.zip)

### How to reproduce

Steps to reproduce the error:
1) open a png layer file that won't insert with action->insert layers
2) check that it is displayed ok
3) hit "save as" to have it in a compatible. Et voilà !

I get some .tmp files though

### Files

[VAT - 84 x84 mm _ Grids Phrozen Aqua 8K.png.zip](https://github.com/sn4k3/UVtools/files/12576392/VAT.-.84.x84.mm._.Grids.Phrozen.Aqua.8K.png.zip)
[Phrozen_Mini8K_AQ8K_750umVAT4DeepSU8_00h10m_1ml_copy.ctb.zip](https://github.com/sn4k3/UVtools/files/12576395/Phrozen_Mini8K_AQ8K_750umVAT4DeepSU8_00h10m_1ml_copy.ctb.zip)

sn4k3, 2023-09-15:
Will be fixed on next release.

sn4k3, 2023-09-17:
Fixed

closed 2023-09-17 by sn4k3