sn4k3/UVtools #756: [Bug] Suction cup resolution is not sucessful

gageoleighton, 2023-09-12:
### System

```Text
UVtools v3.15.1 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-7700K CPU @ 4.20GHz
Processor cores: 8
Memory RAM: 12.76 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.18
AvaloniaUI: 0.10.21
OpenCV: 4.7.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Test_Model_XP_Finder_10_n_2.6s.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: N/A
- Slicer: Lychee 5.2.201

### Description of the bug

Using the Phrozen XP finder STL Lychee produces a reasonably sliced files.
When loaded into UVTools it correctly detects a suction cup but subsequently "drilling" produces a product that would not ameliorate the problem. i.e. you can repeat the suction cup detection again of the drilled/corrected file without fail.

### How to reproduce

1. Download Phrozen XP from [Here](https://phrozen3d.com/pages/phrozens-xp-finder-and-rp-tester-test-model-download-and-tutorials)
2. Slice with any reasonable slicer. In this case Lychee.
3. Load into UVTools and run issue detection.
4. Detection of blank layer (irrelevant) and single suction cup produced by exterior wall surrounding "detailed" region.
5. Correct using "drilling" in UVTools puts a "hole" directly in the center of the model. This would not be appropriate to correct the suction cup.
6. Due to a failed "correction" you can continue to detect the suction cup indefinitely.

gageoleighton, 2023-09-13:
Addendum.

I want to be clear that the issue is with the program "thinking" that the suction cup has been resolved. If you print this, for example, directly on the plate it will make a suction cup. However, if you use a raft and loose supports then the holes in the model work as vents.
For the attached sliced model, I don't think it can be resolved as coded. Perhaps it can't be resolved at all without supports. Rename the extension as .goo as I couldn't upload here.

Based on the fact that this probably doesn't need to be resolved in UVTools I'm going to preemptively mark this as closed. However, should someone with the desire and coding skills to do this then it could be added as a self check. i.e. run the suction cup detection again and if not resolved tell the user.
[Test_Model_XP_Finder_15_n_2.5s.txt](https://github.com/sn4k3/UVtools/files/12591890/Test_Model_XP_Finder_15_n_2.5s.txt)

closed 2023-09-13 by gageoleighton

sn4k3, 2023-09-16:
Suction cups and islands fix is an attempt, it does not guarantee it correctness nor success. If issue goes down to plate it won't drill and will keep detecting. Additionally you should not try to fix stress tests or any other tests that are design to print as is it.