sn4k3/UVtools #760: [FEATURE REQUEST] There are many Chinese users using it now, hoping to add Chinese language

xxbetta, 2023-09-22:
### Describe the solution you'd like

There are many Chinese users using it now, hoping to add Chinese language
3Q

RatmanL, 2026-09-29:
Hi @sn4k3, I would like to add another use case to this language-support request. I have published an unofficial, AI-assisted Czech localization prototype based on UVtools 7.0.0: https://github.com/RatmanL/UVtools-CZ

The fork includes an editable Czech dictionary, source changes, build instructions, tests and screenshots. I built and checked it on CachyOS/Arch Linux x64. It is a proof of concept, not an upstream-ready localization system: it matches English display strings, keeps Czech always active, and some technical or dynamic text remains untranslated. No physical printer compatibility is claimed.

Would you be open to a built-in way to load language packs? Stable text keys, one resource file per language, a language selector and an English fallback would let communities contribute Czech, Chinese and other translations without maintaining separate forks. Is there a localization approach you would prefer for UVtools? I can adapt the Czech dictionary and help test an upstream-friendly design.

Thank you for your work on UVtools!

RatmanL

sn4k3, 2026-09-29:
My project [NetSonar](https://github.com/sn4k3/NetSonar/tree/main/src/NetSonar/Localization) have a localization service built arround Avalonia, we can use that, translating UVtools is not a easy task