sn4k3/UVtools #762: [Bug]  4.0.1 Crashing frequently on OS X

veddermatic, 2023-10-01:
### System

```Text
UVtools v4.0.1 Arm64
Operative system: Darwin 22.6.0 Darwin Kernel Version 22.6.0: Wed Jul  5 22:22:05 PDT 2023; root:xnu-8796.141.3~6/RELEASE_ARM64_T6000 Arm64
Processor: Apple M1 Pro
Processor cores: 8
Memory RAM: 5.48 / 10.10 GB
Runtime: osx.13-arm64
Framework: .NET 6.0.22
AvaloniaUI: 11.0.4
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 1512 x 982 @ 100% (Primary)
    WA: 1512 x 944    UA: 1512 x 944

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Mono 4k
- Slicer:  Lychee

### Description of the bug

Since upgrading to 4.0.1 UV Tools crashes more often than not when running issue detection.

### How to reproduce

1. Be on OS X 13.5.2
2. Open a .pwma file
3. Detect Issues
4. Cross fingers...... oh, it crashed =(

Seems not to be file dependent. I can open a file, detect, get the crash, then reopen UV Tools, open that file again and have it detect fine.

sn4k3, 2023-10-02:
This bug is outside UVtools scope, it comes from UI framework. I will report the bug to them.
Please stick with the older version until this is fixed.

sn4k3, 2023-10-03:
@timunie The last older version it use Avalonia 0.10.21, but even other past version's were ok. This is the first report I saw about this problem.

sn4k3, 2023-10-03:
@veddermatic Please try this compilation:

1. Download: https://we.tl/t-z24JEydM6s
2. Extract the app to your Desktop
3. Open terminal and run: `codesign --force --deep --sign - $HOME/Desktop/UVtools.app`
4. Open UVtools from desktop, try and report back

For the reference, it was compiled with: [image]

sn4k3, 2023-10-19:
Should be fixed with new version, please check.

closed 2023-10-27 by sn4k3