sn4k3/UVtools #763: [Bug] Wrong retract heights on .goo file

mavma, 2023-10-01:
### System

```Text
UVtools v4.0.1 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 5 5600X 6-Core Processor             
Processor cores: 12
Memory RAM: 10.11 / 15.93 GB
Runtime: win10-x64
Framework: .NET 6.0.22
AvaloniaUI: 11.0.4
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 3440 x 1440 @ 125% (Primary) (On this)
    WA: 3440 x 1380    UA: 2752 x 1104

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: rook_u.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Elegoo Mars 4 Max
- Slicer: Lychee

### Description of the bug

After using UVTools to change the lift height, the total lift height and total retract height on the file do not match. 

On the UVTools interface the retract numbers are correctly adjusted, but not on the resulting file. This causes the plate to either not retract the whole way (if lift height was increased), or crash into the bottom (if it was decreased).

### How to reproduce

Example with presliced Elegoo rook (from https://eu.elegoo.com/blogs/3d-printing/elegoo-mars-4-max-3d-printer-support-files). Original settings:
Lift Height 3.00 > 4.00 mm
Retract Height: 5.50 > 1.50 mm

1. Open goo file
2. Tools > Edit print parameters
3. Change Lift Height to 3.00 > 8.00 mm
4. Click "Edit print parameters" button and confirm the change. On the UVTools GUI, the Retract Height correctly changes to 9.50 > 1.50 mm
5. Close the Edit print parameters dialog
6. File > Save as
7. Inspect the resulting .goo file directly. Lift Height in the file will be changed, but Retract Height will still be the original 5.50 > 1.50 mm. If the file is loaded on the printer, the build plate will lift 11mm, but retract only 7mm.

sn4k3, 2023-10-01:
Will fix on next patch

sn4k3, 2023-10-02:
Fixed

closed 2023-10-02 by sn4k3