sn4k3/UVtools #767: [Bug]

BblackmilkK, 2023-10-10:
### System

```Text
UVtools v4.0.2 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Processor cores: 12
Memory RAM: 11.76 / 31.94 GB
Runtime: win10-x64
Framework: .NET 6.0.22
AvaloniaUI: 11.0.4
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1392    UA: 2560 x 1392

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 3lu x.pm3m [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: m3 max / sonic mega 8k / mono x
- Slicer: chitubox pro

### Description of the bug

There are 2 different dimensions. What is this ?

### How to reproduce

re install (clean)

### Files

[image]
[image]

BblackmilkK, 2023-10-10:
[image]
[image]

sn4k3, 2023-10-10:
I don't get it, what is exactly the question?

closed 2023-11-09 by sn4k3