sn4k3/UVtools #773: [Bug] Crashing Linux after import

comato, 2023-10-20:
### System

```Text
UVtools v4.0.3 X64
Operative system: Linux 5.15.0-86-generic #96-Ubuntu SMP Wed Sep 20 08:23:49 UTC 2023 X64
Processor: Intel(R) Core(TM) i5-4670K CPU @ 3.40GHz
Processor cores: 4
Memory RAM: 9.66 / 15.56 GB
Runtime: ubuntu.22.04-x64
Framework: .NET 6.0.23
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary)
    WA: 3840 x 2110    UA: 2560 x 1407

Path:       /tmp/.mount_UVtoolACMrUa/usr/bin/
Executable: /tmp/.mount_UVtoolACMrUa/usr/bin/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

If I import any file... open zip for PCB exposure, configure import, select gerber... it is working, but after click to import button, linux is crashed - black screen, and after some soconds it show linux login screen. It is same, if I want to import png to new layer. I can open file, configure import, and after click to import, it is crashed.
It is same on v 4.0.3 and older v3.15..

In dmesg I found:
[ 3925.800524] show_signal: 10 callbacks suppressed
[ 3925.800527] traps: UVtools[8443] general protection fault ip:7fdc755b03fc sp:7fdbc1dfe3f0 error:0 in libc.so.6[7fdc75593000+195000]
[ 4049.737657] UVtools[9212]: segfault at 8 ip 00007fdeec81227c sp 00007fdee5008530 error 6 in libcoreclr.so[7fdeec6d0000+254000]
[ 4049.737669] Code: 2a 7e 24 00 c3 66 0f 1f 84 00 00 00 00 00 55 48 89 e5 40 84 ff 74 18 66 48 8d 3d 2f 4e 21 00 66 66 48 e8 67 ef ba ff 48 8b 00 <f0> 83 48 08 40 48 8d 05 40 8f 23 00 48 8b 38 48 85 ff 74 09 48 8b
[ 4050.234718] resource sanity check: requesting [mem 0x000c0000-0x000fffff], which spans more than PCI Bus 0000:00 [mem 0x000d0000-0x000e7fff window]
[ 4050.234724] caller os_map_kernel_space.part.0+0x83/0xb0 [nvidia] mapping multiple BARs
[ 4287.937035] .NET ThreadPool[9879]: segfault at 7febf44f5e6b ip 00007fef47832327 sp 00007feed4f9e6f8 error 4 in libICE.so.6.3.0[7fef4782e000+f000]
[ 4287.937047] Code: 48 83 c4 08 5b 5d c3 66 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 8b 07 ff 60 50 66 0f 1f 44 00 00 f3 0f 1e fa 48 8b 07 <ff> 60 58 66 0f 1f 44 00 00 f3 0f 1e fa 48 8b 07 ff 60 60 66 0f 1f

### How to reproduce

Go to import new layer, open png file, or any... and click to import button.

sn4k3, 2023-10-21:
Which grafic card you have?

comato, 2023-10-23:
Palit GTX1060 with default settings, no overclocked.
I try it also on laptop (same Kubuntu), there is Intel Iris 540, it is works ok

sn4k3, 2023-10-23:
Have you the correct Nvidia driver installed? The crash informs the GPU PCI bus is trying to access more memory than it have

comato, 2023-10-23:
I had driver v470, I updated  it to v535 (with lot of issues with network, sound card.. in kernel..). Now it is working correctly. Thank you for navigation..

closed 2023-10-23 by comato