sn4k3/UVtools #775: [Bug] Breaks m5s .pm5s files

aWildTomAppeared, 2023-10-27:
### System

```Text
UVtools v4.0.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i5-10600K CPU @ 4.10GHz
Processor cores: 12
Memory RAM: 21.98 / 31.88 GB
Runtime: win10-x64
Framework: .NET 6.0.22
AvaloniaUI: 11.0.4
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary) (On this)
    WA: 2560 x 1400    UA: 2560 x 1400
2: 2160 x 3840 @ 150%
    WA: 2160 x 3780    UA: 1440 x 2520
3: 1920 x 1080 @ 100%
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Users\Thomas B-B\My_Documents\Software\3D Printing\UVtools_win-x64_v4.0.2\
Executable: C:\Users\Thomas B-B\My_Documents\Software\3D Printing\UVtools_win-x64_v4.0.2\UVtools.exe
Loaded file: cones.pm5s [Version: 518] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Anycubic M5s
- Slicer: Lychee
I've tried this with 3.15 and 4.0.2, I don't have the m5s, it belongs to someone I was trying to help print clear resin better with dimming methods, but the prints didn't work. They tried to slice with 4.0.3 and regardless of the version of which one of us tried it didn't work. 

### Description of the bug

Even if no changes are made to a .pm5s file in UVT, it's just opened and saved then the new file is a bit smaller and it doesn't print. After the transition layers the screen is black, sometimes it never moves to another layer, sometimes it flashes the layers in a fraction of a seconds, it just doesn't work past the transition layers. I tried to fix this by making every layer a transition layer and then manually changing the per layer properties so that they had normal exposures and that did let the layers display but they all had the transition layer exposures not the exposure I manually set. per-layer

### How to reproduce

Open any .pm5s file with UVT, save it, then run it. I can get you in contact with my friend with the pm5s so he can run tests for you if you would like. 

### Files

File before saving from UVT:
https://1drv.ms/u/s!AgdLT8UsgBfshwjDrJjMNWJZAGon?e=7vrqJ0

File after:
https://1drv.ms/u/s!AgdLT8UsgBfshwltb63Zfd3rfQAs?e=1BeGpp

sn4k3, 2023-10-27:
The problem is caused because UVtools turn the per layer data flag due different value detection global vs layer.
On this post (read whole) you can find the manual solution: https://github.com/sn4k3/UVtools/issues/742

sn4k3, 2023-11-09:
Try the new release, should be fixed.

closed 2023-11-09 by sn4k3