sn4k3/UVtools #776: [Bug] Converting from Prusa Slicer to .prz format cause stall of the printer

ecospline, 2023-10-27:
### System

```Text
UVtools v4.0.3 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Xeon(R) CPU E5-2683 v3 @ 2.00GHz
Processor cores: 28
Memory RAM: 8.74 / 31.92 GB
Runtime: win10-x64
Framework: .NET 6.0.23
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Phrozen Sonic Mini 8K S
- Slicer: Prusa Slicer 2.6.1

### Description of the bug

Conversion of a .sl1 file from Prusa Slicer 2.6.1 to .prz format using specyfic converter cause of stalling the printer at the begining of the print - usually with information "Error! Printing file data exception!"  
[image]

### How to reproduce

1. Prepare file in Prusa Slice 2.6.1 , slice it and export in .sl1 format 
2. Open .sl1 file ion UVTools 
3. Convert the file to .prz format to gain acces to the whole parameters of the print (Edit Parameters in UV tools) 
4. Change parameters of the print
5. Save file (overwrite .prz file) 
6. Save final file on Phrozens USB stick 
7. Try to print the file...

sn4k3, 2023-10-27:
Can provide the file?

sn4k3, 2024-01-19:
Fixed

closed 2024-01-19 by sn4k3