sn4k3/UVtools #784: Dynamic layer height resetting dynamic lifts

Hazoooka, 2023-11-05:
### System

```Text
osx.10.15-x64
win7-64
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

Applying the dynamic layer height is resetting previously applied dynamic lifts.

### How to reproduce

1. apply dynamic lifts
2. apply dynamic layer height
3. check the lifts layer by layer and notice them reset to the initial values

sn4k3, 2023-11-06:
All file properties get rebuilt with global settings. As it will lose layers and possible change bottom and normal layers it's safer to reset the values. 
The dynamic lifts should always be applied as last operation as it depends on mass to get the calculations. 
Operations that change layer mass (Dynamic heights for example) will invalidate previous applied dynamic lifts

Hazoooka, 2023-11-06:
Thank you.

Yes, I do that, the dynamic lifts goes after the layer heights.

Kind regards,
Witold.

closed 2023-11-07 by sn4k3