sn4k3/UVtools #787: [Bug] Anybucic X 6Ks: "File is damaged. Print stop!"

EfogDev, 2023-11-08:
### System

```Text
UVtools v4.0.3 X64
Operative system: Linux 6.5.9-1-MANJARO #1 SMP PREEMPT_DYNAMIC Wed Oct 25 13:14:27 UTC 2023 X64
Processor: AMD Ryzen 7 5700X 8-Core Processor
Processor cores: 16
Memory RAM: 29.28 / 31.23 GB
Runtime: manjaro-x64
Framework: .NET 6.0.23
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary)
    WA: 2560 x 1408    UA: 2560 x 1408
2: 2560 x 1440 @ 100%
    WA: 2560 x 1408    UA: 2560 x 1408

Path:       /tmp/.mount_UVtool3goTFJ/usr/bin/
Executable: /tmp/.mount_UVtool3goTFJ/usr/bin/UVtools
```

### Printer and Slicer

- Printer: Anybucic X 6Ks
- Slicer: Lychee Slicer 5.4.0

### Description of the bug

Unfortunately I'm unable to do an exposure test.
File is seen by the printer and has the correct preview.
The build plate lowers to zero, then the error message appears: "File is damaged. Print stop!"

### How to reproduce

Steps to reproduce:
1) Slice anything with Lychee (haven't tried anything else yet)
2) Open with uvtools and make an exposure calibration test 
3) Save it and try to print

### Files

Original file sliced with lychee: sign_74.px6s
After uvtools: calibration.px6s

[files.zip](https://github.com/sn4k3/UVtools/files/13294592/files.zip)

sn4k3, 2023-11-08:
Problem explained here: https://github.com/sn4k3/UVtools/discussions/782

sn4k3, 2023-11-09:
Fixed

closed 2023-11-09 by sn4k3