sn4k3/UVtools #789: Honeycomb Infill does not completely infill my model [Bug]

Strix786, 2023-11-18:
### System

```Text
UVtools v4.0.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i5-8600K CPU @ 3.60GHz
Processor cores: 6
Memory RAM: 12.46 / 15.92 GB
Runtime: win10-x64
Framework: .NET 6.0.24
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 1920 x 1080 @ 100%
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Users\User1\Downloads\UVtools Win x64 v4.0.4  dt2023-11-17\
Executable: C:\Users\User1\Downloads\UVtools Win x64 v4.0.4  dt2023-11-17\UVtools.exe
Loaded file: RT(DP) - Side - Motor dc2023-11-17 v0.8 [UVtooled].sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Phrozen Mighty 8k
- Slicer: Prusa Slicer 2.5.0

### Description of the bug

The bug manifests in a basic model I created: a rectangle. I encounter an issue when applying a Honeycomb infill pattern with an infill spacing of less than 300 pixels. Specifically, the left portion of my rectangle remains unfilled, while the rest of the area is correctly infilled. This issue appears to be solely dependent on the infill spacing, as altering other parameters did not result in any changes.

Furthermore, the wall thickness is consistently maintained throughout the model. Notably, this bug only occurs with models that have a length of up to 200 mm.

### How to reproduce

Go to Tools > Infill
- Select any layer
- Select Honeycomb pattern
- Set anything for floor/ceiling, wall thickness, infill brightness, infill thickness
- Set Infill spacing to something under 300px
- Tick or not Reinforce if possible

### Files

[Bug Honeycomb UVtools.zip](https://github.com/sn4k3/UVtools/files/13400010/Bug.Honeycomb.UVtools.zip)

sn4k3, 2023-11-18:
I remember the calculated square for infill is rounded up, however something is failing, I will look and try to fix on next patch.

sn4k3, 2023-12-12:
Fixed

closed 2023-12-12 by sn4k3