sn4k3/UVtools #792: [Bug] Pixel arithmetic stops early as soon as there is an enclosed flat space on the layer. (False interpretation of end end-of-job?)

tlhintoq, 2023-11-25:
### System

```Text
UVtools v4.0.4 X64
Operative system: Microsoft Windows 10.0.19044 X64
Graphic card: 
Processor: 11th Gen Intel(R) Core(TM) i9-11900H @ 2.50GHz
Processor cores: 16
Memory RAM: 26.80 / 63.71 GB
Runtime: win10-x64
Framework: .NET 6.0.24
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040
2: 2560 x 1080 @ 100%
    WA: 2560 x 1040    UA: 2560 x 1040
3: 2560 x 1080 @ 100%
    WA: 2560 x 1040    UA: 2560 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: 4xmagCap.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer:  SaturnS
- Slicer: Lychee

### Description of the bug

When applying Pixel Arithmetic to produce a corroded effect, as soon as a layer has a portion of enclosed flat the function stops applying the affect.  Its as if it interprets the flat area as end of model.

### How to reproduce

Take any object that has an enclosed flat area on a layer partway through the job.
Try to apply Pixel arithmetic.
Notice it stops at that layer.
[image]

### Files

[GOLD-201.zip](https://github.com/sn4k3/UVtools/files/13465478/GOLD-201.zip)
Notice on the attached, as soon as the flat engraved background is reach the rest of the job doesn't have the effect around the edges.

sn4k3, 2023-11-26:
Try to set ignore areas to **0**

tlhintoq, 2023-11-26:
> Try to set ignore areas to **0**

I had tried smaller than 2mm, didn't think to try 0.
{testing}
Nope, still does it at the same spot.
[image]

sn4k3, 2023-11-26:
Please send the file on question, as your example did work for me when setting ignore areas to 0

tlhintoq, 2023-11-27:
Feel free to close the bug.
I just realized the cause - but not fully.
For some reason on a job of 2548 layers the Pixel Arithmetic dialog is launching and populating with 0-1533 layers. See picture above.

So it did what it was told: Only corroding to layer 1533.
I don't know why it launched the dialog _not_ automatically selecting the full height - but that's the reason it stopped part way through.
[image]

Maybe 1533 what a layer number from a previous job?  I don't know why it picked that layer randomly.

So... This wasn't a bug - just a user error not catching that the dialog wasn't using all the layers.

tlhintoq, 2023-11-27:
If you want to have a play at figuring out why it is choosing 1533 every time I drag in this .ctb file, its 214mb so would need a place to upload it.

sn4k3, 2023-11-27:
Its because you loading your profile which you set from 1 to 1533 layers 

[image]

To avoid that you must select all layer using the SELECT button and save the profile that way, next time it will always select all layers. If you want to skip first layer you can manual edit the file of profiles and set last layer to a large number, eg: 999999

sn4k3, 2023-11-27:
You may have set to default it by click on "D" button near it. To undefault you need to remove it, then you can add it again, it will lose the default on open.

tlhintoq, 2023-11-27:
Ah - yes.  Thanks for the memory boost.  I totally forgot that, then once you mentioned it - it was like "oh yeah, I did do that".
Total brain fart

sn4k3, 2023-11-27:
So now it's working as expected?

tlhintoq, 2023-11-27:
Yes.  Thank you.  Completely user error that just coincidentally happened at that flat point.

closed 2023-11-27 by sn4k3