sn4k3/UVtools #797: [Crash] System.NullReferenceException: Object reference not set to an instance of an object.

seva-luchianov, 2023-12-10:
### System

```Text
UVtools v4.0.4 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 9 3950X 16-Core Processor            
Processor cores: 32
Memory RAM: 21.56 / 63.95 GB
Runtime: win10-x64
Framework: .NET 6.0.24
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

```
System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Rendering.Composition.Compositor.RequestCompositionUpdate(Action action)
   at Avalonia.Rendering.Composition.CompositingRenderer.AddDirty(Visual visual)
   at Avalonia.Reactive.LightweightObservableBase`1.PublishNext(T value)
   at Avalonia.AvaloniaObject.RaisePropertyChanged[T](AvaloniaProperty`1 property, Optional`1 oldValue, BindingValue`1 newValue, BindingPriority priority, Boolean isEffectiveValue)
   at Avalonia.PropertyStore.EffectiveValue`1.SetAndRaiseCore(ValueStore owner, StyledProperty`1 property, T value, BindingPriority priority, Boolean isOverriddenCurrentValue, Boolean isCoercedDefaultValue)
   at Avalonia.PropertyStore.EffectiveValue`1.SetAndRaise(ValueStore owner, IValueEntry value, BindingPriority priority)
   at Avalonia.PropertyStore.ValueStore.ReevaluateEffectiveValues(IValueEntry changedValueEntry)
   at Avalonia.PropertyStore.ValueStore.OnFrameActivationChanged(ValueFrame frame)
   at Avalonia.Styling.StyleInstance.Avalonia.Styling.Activators.IStyleActivatorSink.OnNext(Boolean value)
   at Avalonia.Styling.Activators.StyleActivatorBase.ReevaluateIsActive()
   at Avalonia.Styling.Activators.AndActivator.Avalonia.Styling.Activators.IStyleActivatorSink.OnNext(Boolean value)
   at Avalonia.Styling.Activators.StyleActivatorBase.ReevaluateIsActive()
   at Avalonia.Styling.Activators.StyleClassActivator.Avalonia.Controls.IClassesChangedListener.Changed()
   at Avalonia.Controls.Classes.NotifyChanged()
   at Avalonia.Controls.Classes.Avalonia.Controls.IPseudoClasses.Remove(String name)
   at Avalonia.Input.InputElement.UpdatePseudoClasses(Nullable`1 isFocused, Nullable`1 isPointerOver)
   at Avalonia.Input.InputElement.OnPropertyChanged(AvaloniaPropertyChangedEventArgs change)
   at Avalonia.Controls.Control.OnPropertyChanged(AvaloniaPropertyChangedEventArgs change)
   at Avalonia.Animation.Animatable.OnPropertyChangedCore(AvaloniaPropertyChangedEventArgs change)
   at Avalonia.Input.InputElement.<>c.<.cctor>b__32_6(InputElement x, PointerEventArgs e)
   at Avalonia.Reactive.LightweightObservableBase`1.PublishNext(T value)
   at Avalonia.Interactivity.EventRoute.RaiseEventImpl(RoutedEventArgs e)
   at Avalonia.Interactivity.EventRoute.RaiseEvent(Interactive source, RoutedEventArgs e)
   at Avalonia.Interactivity.Interactive.RaiseEvent(RoutedEventArgs e)
   at Avalonia.Input.PointerOverPreProcessor.ClearPointerOver(IPointer pointer, IInputRoot root, UInt64 timestamp, Nullable`1 position, PointerPointProperties properties, KeyModifiers inputModifiers)
   at Avalonia.Input.PointerOverPreProcessor.OnNext(RawInputEventArgs value)
   at Avalonia.Reactive.LightweightObservableBase`1.PublishNext(T value)
   at Avalonia.Input.InputManager.ProcessInput(RawInputEventArgs e)
   at Avalonia.Controls.TopLevel.HandleInput(RawInputEventArgs e)
   at Avalonia.Win32.WindowImpl.AppWndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.WindowImpl.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.UI.Program.Main(String[] args) in UVtools.UI\Program.cs:line 174
Category: Application
File: _base_tree.dl2p  [Version: 516] [Class: PhotonWorkshopFile]

Machine date time: 12/9/2023 10:39:34 PM
    UTC date time: 12/10/2023 3:39:34 AM
```

### How to reproduce

Happened after clicking the remove island context menu option in the issues section. Have done it a million times before without issue, not sure what actually triggered it to happen.

sn4k3, 2023-12-10:
Looks like UI framework problem.
If very rare occurrence it will be hard to trace.

sn4k3, 2024-01-17:
Unable to reproduce

closed 2024-01-17 by sn4k3