sn4k3/UVtools #800: [Bug] 'Wait time before cure' settings not applying

mtuckerstaley, 2023-12-11:
### System

```Text
UVtools v4.0.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Xeon(R) E-2186M  CPU @ 2.90GHz
Processor cores: 12
Memory RAM: 16.23 / 63.85 GB
Runtime: win10-x64
Framework: .NET 6.0.24
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 100% (Primary)
    WA: 1920 x 1160    UA: 1920 x 1160
2: 1920 x 1200 @ 100% (On this)
    WA: 1920 x 1160    UA: 1920 x 1160

Path:       C:\Users\stalm008\Desktop\MSt\3D Printing\UVtools_win-x64_v4.0.4\
Executable: C:\Users\stalm008\Desktop\MSt\3D Printing\UVtools_win-x64_v4.0.4\UVtools.exe
Loaded file: shell_3_of_Repulsor_Side_.pm3 [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer: Photon M3
- Slicer: Lychee v5

### Description of the bug

Following directions to eliminate/reduce elephant foot using Wait time before/after cure in UVtools
https://blog.honzamrazek.cz/2022/07/step-by-step-guide-on-perfect-bed-adhesion-and-elephant-foot-removal-in-uvtools-3/
Settings do not flag suggestion for base layers and changes only show on the first wait time slot.
I am using the free version of Lychee, which does not have wait time settings, so wondering if I am not able to apply them using UVtools.

### How to reproduce

Change values for wait times. Wait time before cure - Bottom to 20sec, Normal to 2.5sec
Observe suggestions (only one value is flagged) and only one is applied.  2.5/0/0

### Files

[image]

sn4k3, 2023-12-11:
Because Anycubic machines are very limited and don't have a wait time per layer or per bottoms. Only have a global wait time that affect all layers without distinction and then this method does not apply to Anycubic machines. 
You need to use other methods like dim or erode

mtuckerstaley, 2023-12-11:
Many thanks!
Looking into the double exposure tool as well.

sn4k3, 2023-12-11:
Don't use the double exposure, I doubt it will work with anycubic since it use layers in same position and as I know ac printers are unable to.
Use Morph erode is best for your printer

closed 2023-12-11 by sn4k3