sn4k3/UVtools #802: [Crash] System.NullReferenceException: Object reference not set to an instance of an object.

YazawaMasaki, 2023-12-14:
### System

```Text
UVtools v4.0.4 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
Processor cores: 12
Memory RAM: 25.89 / 63.87 GB
Runtime: win10-x64
Framework: .NET 6.0.24
AvaloniaUI: 11.0.5
OpenCV: 4.8.0

Path:       C:\Users\Admin\Desktop\UVtools_win-x64_v4.0.4\
Executable: C:\Users\Admin\Desktop\UVtools_win-x64_v4.0.4\UVtools.exe
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

Remove an empty name profile from exposure time finder dialog cause crash.

```
System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.Core.Operations.OperationCalibrateExposureFinder.get_Xppmm() in UVtools.Core\Operations\OperationCalibrateExposureFinder.cs:line 380
   at UVtools.Core.Operations.OperationCalibrateExposureFinder.get_Holes() in UVtools.Core\Operations\OperationCalibrateExposureFinder.cs:line 582
   at UVtools.Core.Operations.OperationCalibrateExposureFinder.ToString() in UVtools.Core\Operations\OperationCalibrateExposureFinder.cs:line 344
   at System.Runtime.CompilerServices.DefaultInterpolatedStringHandler.AppendFormatted[T](T value)
   at UVtools.UI.Windows.ToolWindow.RemoveSelectedProfile() in UVtools.UI\Windows\ToolWindow.axaml.cs:line 457
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.SendOrPostCallbackDispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.UI.Program.Main(String[] args) in UVtools.UI\Program.cs:line 174
Category: Application
File: 685-Gray-16-0.25.cxdlpv4  [Version: 4] [Class: CXDLPv4File]

Machine date time: 12/14/2023 11:14:46 AM
    UTC date time: 12/14/2023 3:14:46 AM
```

### How to reproduce

Open any file
Calibration -> Exposure time finder
Select an empty named profile
Click delete
Crash
[image]

YazawaMasaki, 2023-12-14:
Update: Creating new profile at this moment will also cause crash.

sn4k3, 2023-12-14:
I'm not able to reproduce this, either add or remove works for me.
You can try to delete the profile file and see if it solve

YazawaMasaki, 2023-12-14:
> I'm not able to reproduce this, either add or remove works for me. You can try to delete the profile file and see if it solve

Currently I cannot delete or add or make default by clicking button on the dialog, either will cause a crash. Is there any alternative method to remove those presumably corrupted profile?

YazawaMasaki, 2023-12-14:
Another update:
Adding profile with name other than "Default" will not cause crash, the newly added profile can be set default or removed without any issue. Seems like "Default" is causing some confusion in program? However I'm still leaved with 2 blank profile I can't touch.

YazawaMasaki, 2023-12-14:
[image]

sn4k3, 2023-12-14:
You can remove the profiles files. By click on Help - open settings directory. Remove or rename the profiles files. But before that attach the file here so I can test with your settings.

YazawaMasaki, 2023-12-14:
Sure here are the files
[UVtools.zip](https://github.com/sn4k3/UVtools/files/13676135/UVtools.zip)
Delete operation_profiles.xml solve the issue, thanks!

sn4k3, 2023-12-14:
With your profiles I was able to reproduce and to get the problematic line, however I have no clue how that is possible, "fixing" that line produce a corruption ahead. Unsure why. Maybe some bad value on configuration, idk.

closed 2023-12-14 by sn4k3