sn4k3/UVtools #803: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.UnknownMethod: Method Destroy is not implemented on interface org.freedesktop.IBus.Service)

pakhomovaleks, 2023-12-14:
### System

```Text
UVtools v4.0.5 X64
Operative system: Linux 6.2.0-37-generic #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov  2 18:01:13 UTC 2 X64
Processor: Intel(R) Core(TM) i7-7600U CPU @ 2.80GHz
Processor cores: 4
Memory RAM: 7.87 / 15.36 GB
Runtime: ubuntu.22.04-x64
Framework: .NET 6.0.25
AvaloniaUI: 11.0.6
OpenCV: 4.8.1

Path:       /tmp/.mount_UVtoolj8V3v8/usr/bin/
Executable: /tmp/.mount_UVtoolj8V3v8/usr/bin/UVtools
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (org.freedesktop.DBus.Error.UnknownMethod: Method Destroy is not implemented on interface org.freedesktop.IBus.Service)
 ---> Tmds.DBus.Protocol.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method Destroy is not implemented on interface org.freedesktop.IBus.Service
   at Tmds.DBus.Protocol.DBusConnection.MyValueTaskSource`1.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
   at Tmds.DBus.Protocol.DBusConnection.CallMethodAsync(MessageBuffer message)
   at Tmds.DBus.Protocol.Connection.CallMethodAsync(MessageBuffer message)
   --- End of inner exception stack trace ---
Category: Task
File: UVtools_demo_file.sl1  [Version: 0] [Class: SL1File]

Machine date time: 12/14/2023 9:27:50 PM
    UTC date time: 12/14/2023 8:27:50 PM
```

### How to reproduce

Start the tool, press open file, press esc

sn4k3, 2023-12-15:
From UI framwork: https://github.com/AvaloniaUI/Avalonia/issues/13863

JoveToo, 2023-12-30:
This happens upon opening the file, it started with 4.0.5.

sn4k3, 2023-12-30:
4.0.5 upgraded the UI framework version, It's a bug in it. Please use 4.0.4 until this problem is fixed.

commit f0de25c0 (2024-01-17): v4.1.0

- **File formats:**
  - (Add) Emake3D Galaxy 1 (QDT) file format and PrusaSlicer printer profile
  - (Add) `ManufacturingProcess` property to get the manufacturing process used by the file/printer combination
  - (Add) `SupportAntiAliasing` property to get if the file supports antialiasing usage (grey pixels)
  - (Improvement) Goo: On encode image do not use gradient compression when going from grey to black or white
  - (Change) PZR: Disable gradient compression for this file format as it corrupt layer for Phrozen Sonic Mini 8K S (#776, #810, #814)
  - (Fix) Thumbnail text generation  [... cut]
closed 2024-01-17 by sn4k3