sn4k3/UVtools #812: [Bug] PCB Exposure. Invert option does not work.

allpassed, 2023-12-28:
### System

```Text
UVtools v4.0.6 X64
Operative system: Linux 5.19.0-40-generic #41~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 31 16:00:14 UTC 2 X64
Processor: Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
Processor cores: 8
Memory RAM: 10.16 / 15.49 GB
Runtime: neon.22.04-x64
Framework: .NET 6.0.25
AvaloniaUI: 11.0.6
OpenCV: 4.8.1

Path:       /tmp/.mount_UVtoolerX3Pu/usr/bin/
Executable: /tmp/.mount_UVtoolerX3Pu/usr/bin/UVtools
Loaded file: xyzCalibration_cube.goo [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn3 Ultra
- Slicer: 

### Description of the bug

Open PCB Exposure tools. Load a Gerber file. Try to check the 'Invert' checkbox for the Geber file.

### How to reproduce

Open PCB Exposure tools.
Load a Gerber file.
Try to check the 'Invert' checkbox for the Geber file.
In my case the checkbox does not change its state. The preview does not change the content.

### Files

[ncp1450_step-up_converter_v4-F_Cu.zip](https://github.com/sn4k3/UVtools/files/13788106/ncp1450_step-up_converter_v4-F_Cu.zip)

sn4k3, 2023-12-28:
There is some bugs in the UI framework involving controls into the DataGrid. It will be fixed on the next release of it. 

As workaround use the global Invert option.

allpassed, 2023-12-28:
Thank you for your quick response. 
Could you please give me a hint on where the global Invert option is in the tool settings?

sn4k3, 2023-12-28:
[image]

allpassed, 2023-12-28:
Oh, great! Thank you for your help!
Best regards.

closed 2024-01-19 by sn4k3