sn4k3/UVtools #814: [Bug] UVtools file kills printer

shuhie, 2023-12-30:
### System

```Text
UVtools v4.0.6 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jun 21 21:18:39 PDT 2022; root:xnu-6153.141.66~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i7-7920HQ CPU @ 3.10GHz
Processor cores: 8
Memory RAM: 6.12 / 16.55 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.25
AvaloniaUI: 11.0.6
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1680 x 1050 @ 100% (Primary)
    WA: 1630 x 1027    UA: 1630 x 1027

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Base.prz [Version: 0] [Class: GooFile]
```

### Printer and Slicer

- Printer: Mini 8KS
- Slicer: lychee 5

### Description of the bug

Imported .prz file from Lychee into uvtools and removed islands. Saved the repaired file as .prz and tried to print file on the Mini 8KS. Printer dies at layer 33. Cannot stop print, cannot move build plate, have to restart printer. 

### How to reproduce

import .prz file from lychee, remove islands, try to print on mini 8KS

### Files

https://we.tl/t-ZrpDXRnsjF

shuhie, 2023-12-30:
The error seems to be similar to this issue, however his failed on the 1st layer, mine fails on 33rd. 
https://github.com/sn4k3/UVtools/discussions/657

This is a picture of the failed print. This happened with both 4.0.4 and 4.0.6
[image]

sn4k3, 2023-12-30:
Yes is the same problem, need to wait for patch

shuhie, 2023-12-30:
> Yes is the same problem, need to wait for patch

well this DLL help in the meantime? 
https://github.com/sn4k3/UVtools/issues/810#issuecomment-1868719320

sn4k3, 2023-12-30:
> well this DLL help in the meantime?

It will

shuhie, 2023-12-30:
> It will

Good deal, I'll give it a try. I appreciate you guys supporting this tool, it is very useful.

sn4k3, 2024-01-19:
Fixed

closed 2024-01-19 by sn4k3