sn4k3/UVtools #815: [Bug] Error importing layers in v4: "Could not import and fit 1 out of 1 format(s) (Insufficient build space)."

SoldierBulls, 2023-12-30:
### System

```Text
UVtools v4.0.6 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: AMD Ryzen 5 3600 6-Core Processor              
Processor cores: 12
Memory RAM: 11.05 / 15.95 GB
Runtime: win10-x64
Framework: .NET 6.0.25
AvaloniaUI: 11.0.6
OpenCV: 4.8.1
```

### Description of the bug

Several "Import layer" options are failing.

I tried version 3.15.1 and it works properly.

But I'm afraid there is a bug in version 4.

### How to reproduce

Actions > Import layers > select any "Merge" or "Bitwise" option to import layers.

It fails with the message "Could not import and fit 1 out of 1 format(s) (Insufficient build space)."

I suspect that this bug could be affecting the other options too.

sn4k3, 2023-12-30:
Without the two files you trying to merge I can't verify anything.
But message is clear, no space in source file to import others...

Most of the import operations require the target files to have the same resolution as the source.

Is your source file resolution equal to target file resolution?

mazapy, 2023-12-30:
Yes! They have the same resolution. I tried with different files, not matter which one. The same files work on v3. Thank you so much for your response :)

mazapy, 2023-12-30:
In fact, I think that the source code is having some kind of problem reading the .Resolution, and that's why this bug is happening.

sn4k3, 2023-12-30:
Then provide the files so I can check.

One difference between v3 and v4 is that v3 do not return any message in case of not importing, it just ignores the unable files and errors.
Read: https://github.com/sn4k3/UVtools/releases/tag/v4.0.0
`Layer actions - Import layer(s): (Improvement) If any file isn't imported it will show an message informing why`

I would not surprised that v3 is doing the same as v4 but do not inform about it

sn4k3, 2023-12-30:
I just tried your sample of exporting and importing back a layer image, I can confirm there is a problem, the resolution is  calculated but not assigned to the variable, which report an empty resolution (0, 0) from file B. It will be fixed on next patch.

This only affect image files. Importing sliced files are unaffected.

mazapy, 2023-12-31:
Awesome! Thank you for everything!
I hope I can help you in the near future with some features :)

Have a nice day!