sn4k3/UVtools #827: Creality HALOT Mage (cxdlpv4 format files) mirrored horizontaly

beattie, 2024-01-23:
### System

```Text
UVtools v4.1.0 X64
Operative system: Linux 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 X64
Processor: AMD Ryzen 3 3300U with Radeon Vega Mobile Gfx
Processor cores: 4
Memory RAM: 3.83 / 5.69 GB
Runtime: ubuntu.22.04-x64
Framework: .NET 6.0.26
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1024    UA: 1920 x 1024
2: 1920 x 1080 @ 100%
    WA: 1920 x 1024    UA: 1920 x 1024

Path:       /tmp/.mount_UVtoolG12UlF/usr/bin/
Executable: /tmp/.mount_UVtoolG12UlF/usr/bin/UVtools
Loaded file: _Phrozen_XP_Finder_V2-6s.cxdlpv4 [Version: 4] [Class: CXDLPv4File]
```

### Printer and Slicer

Creality HALOT Mage (CL-103L)
Chitubox Basic 1.9.5
and
PrusaSlicer 2.7.1+linux-x64-GTK3

### Description of the bug

Files (that print correctly) imported from chitubox are displayed flipped right to left.
Files imported from prusa-slicer display correctly but print flipped,
Files generated by the Calibration -> Exposure time finder tool print flipped right to left.

### How to reproduce

In UVtools open chitubox_Phrozen_XP_Finder_V2.cxdlpv4. display layer 23 the Text "Better Than..." displays mirrored. This file when printed prints the text cirrectly.
(optionaly) In UVtools open prusa_Phrozen_XP_Finder_V2.cxdlpv4 display layer 18 the Text "Better Than..." displays correctly. This file when printed prints the text mirrored.
In UVtools open prusa_Phrozen_XP_Finder_V2.sl1 display layer 18 the Text "Better Than..." displays correctly. UVtools will convert this file to cxdlpv4 which when printed prints the text mirrored.
UVtools can be used to verify the text fo the converted file.

### Files

[UVtools_issue.zip](https://github.com/sn4k3/UVtools/files/14016932/UVtools_issue.zip)

sn4k3, 2024-01-23:
Can be fixed with this option checked:

[image]

I will include the check on next release.

sn4k3, 2024-01-29:
Fixed

closed 2024-01-29 by sn4k3

beattie, 2024-01-30:
It looks like that setting has no effect. I even tried slicing with the SL1 settings and checked in UVtools. It looks like a problem with PrusaSlicer.

sn4k3, 2024-01-30:
But if you slice with prusa slicer checked vs unchecked the direction of image is different right?
Note that UVtools virtually flip image preview to show you the unfliped image. You need to uncheck Flip button on UVtools (If checked) to see the original image

beattie, 2024-01-30:
I went back an looked more and it may be that I was fooled by the preview and the flip option in UVtools. I will do some more careful experiments.