sn4k3/UVtools #829: [Bug] Dialog windows behave erratically with a multiple monitor setup

p0916, 2024-01-24:
### System

```Text
UVtools v4.1.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 7 7700X 8-Core Processor             
Processor cores: 16
Memory RAM: 13.79 / 31.75 GB
Runtime: win10-x64
Framework: .NET 6.0.26
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 100% (Primary)
    WA: 2560 x 1440    UA: 2560 x 1440
2: 1920 x 1080 @ 100%
    WA: 1920 x 1080    UA: 1920 x 1080
3: 1440 x 2560 @ 100% (On this)
    WA: 1440 x 2560    UA: 1440 x 2560

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: conesv2_elegG2.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Description of the bug

I have a 3 monitor setup. I tried to use uvtools on a secondary monitor. Every dialog window will only open on the monitor marked in windows as the main display, despite being open in a different monitor. It will also open the dialog window maximized to the size of the monitor that uvtools is _actually_ open in.

### How to reproduce

1. Move UVtools to a secondary monitor in portrait mode
2. Open a dialog window.

Shows my monitor setup and the behavior when I open "Exposure Time Finder" with uvtools on my secondary monitor. I can not scroll down past "Object Configuration" because the dialog window is opened to a maximum size of my secondary monitor since that is the monitor uvtools is actually in
[image]

Shows the actual size the "Exposure Time Finder" dialog box is trying to be displayed in my main display
[image]

uvtools should open dialog boxes relative to its current location

sn4k3, 2024-01-24:
I have only two monitors with different resolution, and I can't reproduce this. Moving UVtools to secondary monitor and open a tool window results in correct behavior: Tool window open center in same monitor as UVtools main window.

[image]

Every window is set to spawn at center of its parent(Main window), so what happen with you is unexpected.
However, this is out of UVtools scope, the UI Framework (Avalonia) is who manage the spawns, start positions, scales, etc. While is possible to move the window by code, that is not what we want.
As you can reproduce the bug, I recommend you open the issue here: https://github.com/AvaloniaUI/Avalonia/issues

One thing you can try: Do not maximize the main window, keep it smaller than your monitor resolution, open a tool, does it produce same effect?

For UVtools open in the last used monitor you need to change settings:

[image]

If still have problems, you can unlock windows to able to resize them, and that will fix the cut content when you resize them:

[image]

p0916, 2024-01-24:
Thank you for the quick reply! No matter the settings I change in uvtools, the dialog boxes will only open in which ever monitor is set to "main display" within windows. uvtools, itself, will also not open in the last position it was closed if it was closed outside of the main display. This seems to be a windows and/or Avalonia issue.

My apologies as I was unaware of Avalonia and this appears to be an external bug not directly related to uvtools. I appreciate the help!

sn4k3, 2024-01-24:
Avalonia main window always open on primary monitor or from where it got triggered (If you create a shortcut on that monitor it should spawn on that monitor). UVtools then reposition the main window based on last coordinates (If you have those checkboxes on). After quitting it save window last coordinates and restore on next startup. It work on my setup so I guess the problem is in that configuration. Can you try to position all screens landscape and side by side (Just in windows settings) and try?

However the tool window should spawn in center of its parent, so I really think it wroth to open the issue on their side.
Because it open in other monitor, the tool window gets the size and scale of that monitor. Thats why you need to check the resize option and resize it after all to fix it wrong auto-size.

closed 2024-02-29 by sn4k3