sn4k3/UVtools #830: [Bug] Constant crashes on Linux

tddts, 2024-01-29:
### System

```Text
UVtools v4.1.0 X64
Operative system: Linux 5.15.0-92-generic #102-Ubuntu SMP Wed Jan 10 09:33:48 UTC 2024 X64
Processor: AMD Ryzen 7 7700X 8-Core Processor
Processor cores: 16
Memory RAM: 5.18 / 30.49 GB
Runtime: linuxmint.21.3-x64
Framework: .NET 6.0.26
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       /tmp/.mount_UVtoolQbQSOE/usr/bin/
Executable: /tmp/.mount_UVtoolQbQSOE/usr/bin/UVtools
```

### Printer and Slicer

- Printer: Elegoo Saturn 3 Ultra
- Slicer:  Lychee 5.4.300

### Description of the bug

App crashes whether when opening a file or trying to open "issues" tab. 
The whole system freezes while app is loading/calculating stuff.
There is no error log in "settings folder" and no output if I run it from the terminal.

### How to reproduce

1. Have latest Linux Mint
2. Try opening file
3. Experience crash
4. If it did not crash, try opening "issues" tab, go to 3.

### Files

There are 2 files here.
First one is a simple cube I've created in lychee, it opens fine.
The second file opened once but it crashed after I tried to open "issues" tab, now it doesn't open at all, crashing on load.
https://drive.google.com/drive/folders/1klkspVe7JEjMR90WdmTua67er07E-XOX?usp=sharing

sn4k3, 2024-01-29:
However the insane amount of RAM it shows using makes little sense, because with your core count the calcs aim for 1GB. For that I review the code for Goo and find that I dont dispose the Mat after using it, so object remain alive even if not required. That mean it decode and keep all 12K bitmaps into RAM and only auto release them when all its processed via GC.

I will fix this on next patch. Only affecting goo.

Difference after fix, on larger file:

[image]

tddts, 2024-01-29:
I tried to decrease parallelism to **one**, but it still consumed all of my RAM + 7Gb of swap (which I did not even expect to be used at all with such amount RAM, but now it seems to be too small) and died.
I tried to slice the same model into .ctb and got the same result when opening it.
So I guess its not only about .goo files.

sn4k3, 2024-01-29:
Can you attach the CTB file?

tddts, 2024-01-29:
here:
https://drive.google.com/file/d/1C_aamvHNi6menlZlaaZMfuE_n2rRr9mw/view?usp=sharing

sn4k3, 2024-01-29:
Ok, EncryptedCTB is also affected by same problem, after that I review all other file formats and found that SVGX is also with same problem. It's strange that no-one complained before with CTB, given the big resolutions and high layer count...

sn4k3, 2024-01-29:
Check new version, should be fixed, including the issue detection leak

closed 2024-01-29 by sn4k3

tddts, 2024-01-30:
4.2.0 works fine with previous files.
I tried opening a [bigger file](https://drive.google.com/file/d/1Wy0PBibOcRoFbR8NIHf_Od7krDCOGFOG/view?usp=sharing) (2828 layers, ~624Mb)
With parallelism set to **Auto (-1)** (LZ4 compression) it opens fine, but on island detection it also consumes all RAM and dies.
With parallelism set to **one** on island detection it consumes up to 5.6 Gb of RAM and works normally.
With parallelism = 10 it consumes up to 7 Gb of RAM.
With parallelism set to MAX (16/16) it consumes up to 8.6 Gb of RAM.
I think there is something wrong with **Auto** setting.

sn4k3, 2024-01-30:
That file is huge and using almost all space in 12k, little to no optimization is made in that case.
The detection of issues will escalate the RAM since per operation it needs multiple Mats, in some cases 2 to 4 and on resin traps need to build a cache of Mat's which is not small.

The Auto (-1) parallelism is not handled by the program but the NET framework: [ParallelOptions.MaxDegreeOfParallelism](https://learn.microsoft.com/en-us/dotnet/api/system.threading.tasks.paralleloptions.maxdegreeofparallelism).

Special this lines:

> ... If it is -1, there is no limit on the number of concurrently running operations (with the exception of the [ForEachAsync] (https://learn.microsoft.com/en-us/dotnet/api/system.threading.tasks.parallel.foreachasync?view=net-8.0) method, where -1 means [ProcessorCount](https://learn.microsoft.com/en-us/dotnet/api/system.environment.processorcount?view=net-8.0)).

> When the thread pool's heuristics is unable to determine the right number of threads to use and could end up injecting too many threads. For example, in long-running loop body iterations, the thread pool might not be able to tell the difference between reasonable progress or livelock or deadlock, and might not be able to reclaim threads that were added to improve performance. In this case, you can set the property to ensure that you don't use more than a reasonable number of threads.

So, in that case the framework manages the creation of tasks based on what he thinks is best for performance. For example, if you pause a task, it will keep spawning them because there are an oportunity to process data (Since others are paused). If it's very busy it will wait an oportunity to spawn a new task. So in the end it may spawn more tasks than your CPU core count. In my case I have 32 and it spawn 34 to 36.

But if you choose a number, it will always stay within that limit.

Myself and what I recommend is to use the "!" (Optimal). It will define to your core count less a few to make sure some cores are free to make your system responsive, that way you can use your OS without much lag.
**-1** should be avoid if you active use your PC during UVtools processing, eg: web, videos, working.

Where the results on my system:
[image]

The most memory is taken on Resin traps due the MatCache requirement. If your models are solid or you know they are ok for sure please disable resin trap detection on such files

MasterCATZ, 2024-12-13:
so maybe having 128 cores is working against me then
so maybe I can tune the threads to not go over 512gb RAM now and stop my Ubuntu desktop reloading 
crazy how oom killer is killing everything and not just the UVtools.AppImage that is hogging everything 

so n * bitmaps   , 
how do you convert bitmaps to  memory size  ?  would 12k = say 12gb RAM? per thread ? or is it smaller ?

sn4k3, 2024-12-13:
> so how the heck am i using up all my RAM ?

- What are your files resolution?
- Total ram?

You have a ton of cores, is that 128 the total or it doubles with hyperthreading?
Note that by default UVtools will spawn a task per core, meanwhile if task is late or there are free tasks it will spawn a bit more to compensate. That of course require a lot of ram. Some memory are only free latter (Not just after task ends).
If by any chance there are a memory leak your system will be killed easy.

My advice is to cut workload in half on UVtools settings, start with 1/2, monitor ram and tune from there