sn4k3/UVtools #833: Error while trying compute issues

niscrulez, 2024-02-04:
### System

```Text
UVtools v4.2.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: Intel(R) Core(TM) i5-8600K CPU @ 3.60GHz
Processor cores: 6
Memory RAM: 6.24 / 15.90 GB
Runtime: win10-x64
Framework: .NET 6.0.26
AvaloniaUI: 11.0.7
OpenCV: 4.8.1

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: big2.cws [Version: 0] [Class: CWSFile]
```

### Printer and Slicer

- Printer: whale 2
- Slicer: lychee
[image]

### Description of the bug

System.AggregateException: One or more errors occurred. (OpenCV: 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows)
 ---> Emgu.CV.Util.CvException: OpenCV: 0 <= roi.x && 0 <= roi.width && roi.x + roi.width <= m.cols && 0 <= roi.y && 0 <= roi.height && roi.y + roi.height <= m.rows
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
   at Emgu.CV.MatInvoke.cveMatCreateFromRect(IntPtr mat, Rectangle& roi)
   at UVtools.Core.Extensions.EmguExtensions.Roi(Mat mat, Rectangle roi, EmptyRoiBehaviour emptyRoiBehaviour) in UVtools.Core\Extensions\EmguExtensions.cs:line 794
   at UVtools.Core.Managers.IssueManager.<>c__DisplayClass18_0.<DetectIssues>b__11(Mat[] mats) in UVtools.Core\Managers\IssueManager.cs:line 665
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)

### How to reproduce

[image]

sn4k3, 2024-02-04:
Can provide the file so I can replicate?

niscrulez, 2024-02-05:
> Can provide the file so I can replicate?

Yes. Thank you. You need to delete the .zip from the file name (Shield.part1.rar + Shield.part2.rar).

[Shield.part2.rar.zip](https://github.com/sn4k3/UVtools/files/14160093/Shield.part2.rar.zip)

[Shield.part1.rar.zip](https://github.com/sn4k3/UVtools/files/14160091/Shield.part1.rar.zip)

sn4k3, 2024-02-05:
Your file have wrong resolution information and the last layer is empty. It will create last layer with file resolution instead of image resolution and that is why last layer is different in resolution. If you remove that last layer and run detection it will work.

It will be fixed on next release.

sn4k3, 2024-02-10:
Fixed

closed 2024-02-10 by sn4k3

niscrulez, 2024-02-10:
Thank you so much.